armbian/build #2394: Focal on Odroid N2+: Constantly starts and stops getty on ttyS0

tomek-szczesny, 2020-11-29:
Hello!

I've got some other problem with my system but here's what I found in the meantime:

```
mctom@Tomusiomat-ARM:~$ sudo tail -n10 /var/log/syslog 
Nov 29 22:59:10 localhost systemd[1]: Stopped Serial Getty on ttyS0.
Nov 29 22:59:10 localhost systemd[1]: Started Serial Getty on ttyS0.
Nov 29 22:59:20 localhost systemd[1]: serial-getty@ttyS0.service: Succeeded.
Nov 29 22:59:21 localhost systemd[1]: serial-getty@ttyS0.service: Scheduled restart job, restart counter is at 142.
Nov 29 22:59:21 localhost systemd[1]: Stopped Serial Getty on ttyS0.
Nov 29 22:59:21 localhost systemd[1]: Started Serial Getty on ttyS0.
Nov 29 22:59:31 localhost systemd[1]: serial-getty@ttyS0.service: Succeeded.
Nov 29 22:59:31 localhost systemd[1]: serial-getty@ttyS0.service: Scheduled restart job, restart counter is at 143.
Nov 29 22:59:31 localhost systemd[1]: Stopped Serial Getty on ttyS0.
Nov 29 22:59:31 localhost systemd[1]: Started Serial Getty on ttyS0.
```

```
mctom@Tomusiomat-ARM:~$ sudo tail -n10 /var/log/auth.log
Nov 29 22:59:10 localhost agetty[6238]: /dev/ttyS0: not a tty
Nov 29 22:59:21 localhost agetty[6240]: /dev/ttyS0: not a tty
Nov 29 22:59:31 localhost agetty[6241]: /dev/ttyS0: not a tty
Nov 29 22:59:34 localhost sudo:    mctom : TTY=pts/0 ; PWD=/home/mctom ; USER=root ; COMMAND=/usr/bin/tail -n10 /var/log/syslog
Nov 29 22:59:34 localhost sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 29 22:59:34 localhost sudo: pam_unix(sudo:session): session closed for user root
Nov 29 22:59:41 localhost agetty[6244]: /dev/ttyS0: not a tty
Nov 29 22:59:51 localhost agetty[6249]: /dev/ttyS0: not a tty
Nov 29 23:00:01 localhost sudo:    mctom : TTY=pts/0 ; PWD=/home/mctom ; USER=root ; COMMAND=/usr/bin/tail -n10 /var/log/auth.log
Nov 29 23:00:01 localhost sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
```

```
mctom@Tomusiomat-ARM:~$ sudo tail -n10 /var/log/Xorg.0.log
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
[  1901.966] (EE) FBDEV(0): FBIOPUTCMAP: Invalid argument
```
(there's actually hundreds of these every second, the logfile swells very quickly)

I realize now that every time I google armbian-related things I'll end up finding Igor. ;)

Apparently [this](https://github.com/armbian/build/commit/3092295cb0513ea12d40ad5aa1d988b09945ec05) was a solution for a similar problem described [here](https://forum.armbian.com/topic/6566-le-potato-serial-getty-on-ttys0-starts-stops-restarts/?tab=comments#comment-49993).

Unfortunately, I don't know how to fix it myself, however I'll be happy to assist with more information. Thanks!

EvilOlaf, 2020-11-30:
Hey,
this issue tracker is for problems you run into while *building* Armbian.
Problems while running Armbian on your board should be reported to forums.
Also include `armbianmonitor -u` when creating your report. This will provide us with important information about your board and used Armbian version.

tomek-szczesny, 2020-11-30:
> this issue tracker is for problems you run into while _building_ Armbian.

Well, yes, but the proposed solution for the problem is to modify a file from this repository. What's the better place to post this? A forum?

igorpecovnik, 2020-11-30:
Already prepared: https://github.com/armbian/build/pull/2389

closed 2020-11-30 by igorpecovnik