armbian/build #2398: [armbian-firstlogin] Provide an option to skip/exit

ashthespy, 2020-11-30:
It looks like with there was a regression with https://github.com/armbian/build/pull/2190
Previously there was an option to skip setting additional languages, but now you are forced to select something. 

https://github.com/armbian/build/blob/301467b172500d6a123c31eda4e04ba2d6ecfb5f/packages/bsp/common/usr/lib/armbian/armbian-firstlogin#L63-L65

Similar case for the root password - would it be possible to add a break to the loop?

igorpecovnik, 2020-11-30:
> Previously there was an option to skip setting additional languages

Also @guidol70 reported one strange outcome https://forum.armbian.com/topic/16112-2011-orange-pi-zero-not-booting/?tab=comments#comment-113750

Variants:

1. X timezone detected. Do you want to set locales based on your location?

- Yes (proceeds as now)
- No (skips creating locales)

2. X location detected. Do you want to choose another location? 

- Yes (invoke TZ selector)
- No (proceeds as now)

3. ?

> Similar case for the root password - would it be possible to add a break to the loop?

New root password (CTRL C to break this process)

ashthespy, 2020-11-30:
From a quick look at #2190, it looks like the old behaviour of "X timezone(s) detected" has been removed with the refactor.
The locale check now assumes if there are multiple locales, then you have to pick one. 
It would be nice to be able to skip this check even before the call to API? 

https://github.com/armbian/build/blob/301467b172500d6a123c31eda4e04ba2d6ecfb5f/packages/bsp/common/usr/lib/armbian/armbian-firstlogin#L63-L76

IIRC, the root password change was always mandatory, but you could just set a dummy one to get on with debugging, but now with `cracklib-check` things get a bit more tedious. 

The add user is still skippable. 
https://github.com/armbian/build/blob/301467b172500d6a123c31eda4e04ba2d6ecfb5f/packages/bsp/common/usr/lib/armbian/armbian-firstlogin#L247-L252

TRSx80, 2020-12-01:
I came here to help out (this look like a relatively easy issue), but it looks like @ashthespy is already working on it, so I won't touch anything for now.

> things get a bit more tedious

Tedious is a very good word, this jumped out to me as well when I had re-installed something recently.

igorpecovnik, 2020-12-02:
> but you could just set a dummy one to get on with debugging

One idea for debugging purposes - if you do this often enough - https://github.com/armbian/autotests

./go.sh PREPAREONLY=yes # prepares all hosts in subnet - go over 1st login, updates, ... 
./go.sh PREPAREONLY=yes HOSTS="xx.xx.xx.xx" # do the same for just some host

heisath, 2020-12-28:
As this is still annoying  and @ashthespy has not provided a fix, maybe you (@TRSx80) want to give it a go?

TRSx80, 2020-12-30:
I would love to but I realized several days (a week?) or so ago that my time to commit to outside projects is basically over for the time being.  I am strictly trying to wrap up some of my own (personal / home) stuff right now and then I will probably disappear for some months as I head back to (paying) work.  Sorry guise.

closed 2021-01-26 by igorpecovnik