armbian/build #2416: build failure with empty container - key error?

5kft, 2020-12-04:
Hi @igorpecovnik , I just purged my docker images as per normal to start with a fresh image, and I'm receiving a gpg key error when doing the build (./compile.sh docker ...):

Output from the command line:
```
[...]
Get:28 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [876 kB]
Get:29 http://archive.ubuntu.com/ubuntu focal-updates/multiverse i386 Packages [6,526 B]
Get:30 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [874 kB]
Get:31 http://archive.ubuntu.com/ubuntu focal-backports/universe i386 Packages [3,031 B]
Get:32 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [4,248 B]
Fetched 25.5 MB in 5s (5,100 kB/s)
Reading package lists...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libjq1 libonig5
The following NEW packages will be installed:
  jq libjq1 libonig5
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 313 kB of archives.
After this operation, 1,062 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal/universe amd64 libonig5 amd64 6.9.4-1 [142 kB]
Get:2 http://archive.ubuntu.com/ubuntu focal/universe amd64 libjq1 amd64 1.6-1 [121 kB]
Get:3 http://archive.ubuntu.com/ubuntu focal/universe amd64 jq amd64 1.6-1 [50.2 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 313 kB in 1s (253 kB/s)
                               Selecting previously unselected package libonig5:amd64.
(Reading database ... 27545 files and directories currently installed.)
Preparing to unpack .../libonig5_6.9.4-1_amd64.deb ...
Unpacking libonig5:amd64 (6.9.4-1) ...
Selecting previously unselected package libjq1:amd64.
Preparing to unpack .../libjq1_1.6-1_amd64.deb ...
Unpacking libjq1:amd64 (1.6-1) ...
Selecting previously unselected package jq.
Preparing to unpack .../archives/jq_1.6-1_amd64.deb ...
Unpacking jq (1.6-1) ...
Setting up libonig5:amd64 (6.9.4-1) ...
Setting up libjq1:amd64 (1.6-1) ...
Setting up jq (1.6-1) ...
Processing triggers for libc-bin (2.31-0ubuntu9.1) ...
[ o.k. ] Checking for external GCC compilers
[ error ] ERROR in function download_and_verify [ general.sh:1310 ]
[ error ] verification failed
[ o.k. ] Process terminated
```

Output from output/debug/output.log:

```
Displaying message: Preparing host info
Displaying message: Build host OS release focal info
Displaying message: Running in container docker info
Displaying message: apt-cacher is disabled in containers, set NO_APT_CACHER=no to override  wrn
Displaying message: Installing build dependencies
Displaying message: Checking for external GCC compilers  info
gpg: keybox '/root/armbian/cache/.gpg/pubring.kbx' created
gpg: /root/armbian/cache/.gpg/trustdb.gpg: trustdb created
gpg: error reading key: No public key
gpg: key D47877808F427EAF: public key "Linaro Toolchain Builder <michael.hope+cbuild@linaro.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: error reading key: No public key
gpg: key 93D6889F9F0E78D5: public key "Igor Pecovnik <igor@armbian.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: no signed data
gpg: can't hash datafile: No data
Displaying message: ERROR in function download_and_verify general.sh:1310 err
Displaying message: verification failed  err
Displaying message: Process terminated  info
```

Any ideas?  This process used to work for me...  Thanks!

5kft, 2020-12-04:
I just did a quick test and reverted commit 76aad7802c59c5ef7635c04fefda638341adf4d7 locally, and the build and download process is working fine again:
```
commit 76aad7802c59c5ef7635c04fefda638341adf4d7
Author: lanefu <lanefu@users.noreply.github.com>
Date:   Tue Dec 1 08:43:38 2020 -0500

    Use armbian mirrors for downloads (#2400)

    * switch to ARBIAN_MIRROR var for downloads
    Co-authored-by: Igor Pečovnik <igorpecovnik@users.noreply.github.com>
```

Command line output:
```
[ warn ] This script requires root privileges, trying to use sudo
[ o.k. ] Using config file [ /home/5kft/projects/armbian/build/userpatches/config-docker.conf ]
[ .... ] Using existed a armbian Docker container
[ o.k. ] Running the container
[ o.k. ] Using config file [ /root/armbian/userpatches/config-example.conf ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ yes ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ yes ]
[ o.k. ] Command line: setting KERNEL_KEEP_CONFIG to [ no ]
[ o.k. ] Command line: setting BUILD_DESKTOP to [ no ]
[ o.k. ] Command line: setting USE_MAINLINE_GOOGLE_MIRROR to [ no ]
[ o.k. ] Command line: setting USE_GITHUB_UBOOT_MIRROR to [ yes ]
[ o.k. ] Command line: setting BUILD_KSRC to [ no ]
[ o.k. ] Command line: setting KERNEL_EXPORT_DEFCONFIG to [ yes ]
[ o.k. ] Command line: setting BRANCH to [ dev ]
[ o.k. ] Command line: setting CREATE_PATCHES to [ no ]
[ o.k. ] Command line: setting BOARD to [ nanopineo2black ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ focal ]
[ o.k. ] Running in container [ docker ]
[ warn ] apt-cacher is disabled in containers, set NO_APT_CACHER=no to override
[ o.k. ] Checking for external GCC compilers
ln: failed to create symbolic link '/root/armbian/cache/toolchain/gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz.asc': File exists
[ .... ] downloading using http(s) network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]
[#af5a0c 23MiB/24MiB(95%) CN:1 DL:2.0MiB]
[ o.k. ] Verified [ PGP ]
[ .... ] decompressing
[ .... ] gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz: 24.9MiB [15.1MiB/s] [========================================================================================>] 100%
[ .... ] downloading using http(s) network [ gcc-linaro-arm-none-eabi-4.8-2014.04_linux.tar.xz ]
[#866ba6 33MiB/33MiB(99%) CN:1 DL:719KiB]
[ o.k. ] Verified [ PGP ]
[ .... ] decompressing
[ .... ] gcc-linaro-arm-none-eabi-4.8-2014.04_linux.tar.xz: 33.9MiB [13.7MiB/s] [===========================================================================================>] 100%
[ .... ] downloading using http(s) network [ gcc-linaro-arm-linux-gnueabihf-4.8-2014.04_linux.tar.xz ]
[#b46402 20MiB/48MiB(42%) CN:1 DL:1.1MiB ETA:24s]
```

Seems like there is an issue with this particular change?

igorpecovnik, 2020-12-06:
Works for me without reverting that. Can you try again.

5kft, 2020-12-06:
Cleared the docker images and volumes, and tried again, and it works now:
```
Preparing to unpack .../libjq1_1.6-1_amd64.deb ...
Unpacking libjq1:amd64 (1.6-1) ...
Selecting previously unselected package jq.
Preparing to unpack .../archives/jq_1.6-1_amd64.deb ...
Unpacking jq (1.6-1) ...
Setting up libonig5:amd64 (6.9.4-1) ...
Setting up libjq1:amd64 (1.6-1) ...
Setting up jq (1.6-1) ...
Processing triggers for libc-bin (2.31-0ubuntu9.1) ...
[ o.k. ] Checking for external GCC compilers
[ .... ] downloading from redirect.armbian.com using http(s) network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]
[#99ffaf 23MiB/24MiB(95%) CN:1 DL:1.7MiB]
[ o.k. ] Verified [ PGP ]
[ .... ] decompressing
[ .... ] gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz: 24.9MiB [14.9MiB/s] [=======================================================>] 100%
[ .... ] downloading from redirect.armbian.com using http(s) network [ gcc-linaro-arm-none-eabi-4.8-2014.04_linux.tar.xz ]
[#b4dbba 2.5MiB/33MiB(7%) CN:1 DL:760KiB ETA:42s]
```
Not sure why, but perhaps there was a caching issue on a mirror somewhere, perhaps?  In any case, it is working now - thanks!

closed 2020-12-06 by 5kft