armbian/build #2445: minio.k-space.ee has an AAAA/IPv6 address which isn't responding

yaleman, 2020-12-13:
I know this isn't specifically covered by armbian build but it's breaking the ability to use it - the package host has an AAAA (IPv6) record, as set up in [#605](https://github.com/armbian/build/issues/605) - but it's stopped working.

## IPv4 works

```
$ curl -vvI4 https://minio.k-space.ee/armbian/apt/dists/buster/InRelease 2>&1 | grep -iE 'HTTP|Failed'
* ALPN, offering http/1.1
> HEAD /armbian/apt/dists/buster/InRelease HTTP/1.1
< HTTP/1.1 200 OK
HTTP/1.1 200 OK
```

## IPv6 Fails

```
$ curl -vvI6 https://minio.k-space.ee/armbian/apt/dists/buster/InRelease 2>&1 | grep -iE 'HTTP|Failed'
  0     0    0     0    0     0      0      0 --:--:--  0:01:16 --:--:--     0* Connection failed
* connect to 2001:bb8:4008:20:193:40:103:4 port 443 failed: Operation timed out
* Failed to connect to minio.k-space.ee port 443: Operation timed out
curl: (7) Failed to connect to minio.k-space.ee port 443: Operation timed out
```

## IPv6 test to Google works (just to validate it's not my routing! :) )

```
$ curl -vvI6 https://google.com 2>&1 | grep -iE 'HTTP|Connected|Failed'
* Connected to google.com (2404:6800:4006:80b::200e) port 443 (#0)
* ALPN, offering http/1.1
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
> HEAD / HTTP/2
< HTTP/2 301
< location: https://www.google.com/
HTTP/2 301
location: https://www.google.com/
```

igorpecovnik, 2020-12-13:
@laurivosandi

igorpecovnik, 2020-12-15:
Fixed.

curl -vv6 -I https://minio.k-space.ee/armbian/apt/dists/buster/InRelease

```
*   Trying 2001:bb8:4008:20:193:40:103:4:443...
* TCP_NODELAY set
* Connected to minio.k-space.ee (2001:bb8:4008:20:193:40:103:4) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: CN=*.k-space.ee
*  start date: Nov 30 18:21:09 2020 GMT
*  expire date: Feb 28 18:21:09 2021 GMT
*  subjectAltName: host "minio.k-space.ee" matched cert's "*.k-space.ee"
*  issuer: C=US; O=Let's Encrypt; CN=Let's Encrypt Authority X3
*  SSL certificate verify ok.

```

closed 2020-12-15 by igorpecovnik