armbian/build #2450: Add 'usb-storage.quirks=${usbstoragequirks}' to Odroid N2 /boot/boot.ini

matty-fb, 2020-12-13:
Hello,

Would it be possible to add `usb-storage.quirks=${usbstoragequirks}` to the Odroid N2 bootargs ([/boot/boot.ini](https://github.com/armbian/build/blob/master/config/bootscripts/boot-odroid-n2.ini#L113) line 113)? The N2 picks up the quirks from amrbianEnv.txt when this is added manually, but boot.ini gets overwritten after upgrades.

Thanks in advance!
Matt

igorpecovnik, 2020-12-14:
Easier and probably best way is to remove this line https://github.com/armbian/build/blob/master/config/boards/odroidn2.conf#L11

matty-fb, 2020-12-15:
Wonderful, thank you for pointing me in the right direction - much appreciated.

matty-fb, 2021-10-22:
Thank you for all of the fantastic work, Igor!