armbian/build #2506: Using caching proxy causes apt.armbian.com to throw 500 errors

yaleman, 2020-12-28:
Configuring a (squid) proxy in line with apt has started to cause it to fail on the redirection-to-the-mirror step.

# Via Proxy

	# curl -vx http://proxy.internal:3128 http://apt.armbian.com/dists/buster/InRelease

	* Connected to proxy.internal (10.0.0.27) port 3128 (#0)
	> GET http://apt.armbian.com/dists/buster/InRelease HTTP/1.1
	> Host: apt.armbian.com
	> User-Agent: curl/7.64.0
	> Accept: */*
	> Proxy-Connection: Keep-Alive
	>
	< HTTP/1.1 500 Internal Server Error
	< Server: nginx/1.18.0 (Ubuntu)
	< Date: Mon, 28 Dec 2020 03:17:35 GMT
	< Content-Type: text/html; charset=utf-8
	< Content-Length: 290
	< X-Cache: MISS from proxy
	< X-Cache-Lookup: MISS from proxy:3128
	< Connection: keep-alive
	<
	<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
	<title>500 Internal Server Error</title>
	<h1>Internal Server Error</h1>
	<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
	* Connection #0 to host proxy.internal left intact
	
# Direct

	# curl -v http://apt.armbian.com/dists/buster/InRelease
	* Connected to apt.armbian.com (2605:7900:20::5) port 80 (#0)
	> GET /dists/buster/InRelease HTTP/1.1
	> Host: apt.armbian.com
	> User-Agent: curl/7.64.0
	> Accept: */*
	>
	< HTTP/1.1 302 FOUND
	< Server: nginx/1.18.0 (Ubuntu)
	< Date: Mon, 28 Dec 2020 03:28:11 GMT
	< Content-Type: text/html; charset=utf-8
	< Content-Length: 329
	< Connection: keep-alive
	< Location: https://mirrors.dotsrc.org/armbian-apt/dists/buster/InRelease
	<
	<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
	<title>Redirecting...</title>
	<h1>Redirecting...</h1>
	* Connection #0 to host apt.armbian.com left intact
	<p>You should be redirected automatically to target URL: <a href="https://mirrors.dotsrc.org/armbian-apt/dists/buster/InRelease">https://mirrors.dotsrc.org/armbian-apt/dists/buster/InRelease</a>.  If not click the link.

# apt configured with proxy

	# sudo apt-get update
	[sudo] password for yaleman:
	Hit:1 http://ftp.au.debian.org/debian buster InRelease
	Hit:2 http://ftp.au.debian.org/debian buster-updates InRelease
	Hit:3 http://security.debian.org buster/updates InRelease
	Err:4 http://apt.armbian.com buster InRelease
	  500  Internal Server Error [IP: 10.0.0.27 3128]
	Hit:5 http://ftp.au.debian.org/debian buster-backports InRelease
	Reading package lists... Done
	W: Failed to fetch http://apt.armbian.com/dists/buster/InRelease  500  Internal Server Error [IP: 10.0.0.27 3128]
	W: Some index files failed to download. They have been ignored, or old ones used instead.
	#

# apt configured without proxy

	$ sudo apt-get update
	Hit:1 http://security.debian.org buster/updates InRelease
	Hit:2 http://ftp.au.debian.org/debian buster InRelease
	Hit:3 http://ftp.au.debian.org/debian buster-updates InRelease
	Hit:4 http://ftp.au.debian.org/debian buster-backports InRelease
	Get:5 https://us.mirrors.fossho.st/armbian/apt buster InRelease [18.3 kB]
	Get:6 https://us.mirrors.fossho.st/armbian/apt buster/main armhf Packages [646 kB]
	Get:7 https://us.mirrors.fossho.st/armbian/apt buster/main arm64 Packages [762 kB]
	Get:8 https://us.mirrors.fossho.st/armbian/apt buster/main arm64 Contents (deb) [11.2 MB]
	Get:9 https://us.mirrors.fossho.st/armbian/apt buster/main armhf Contents (deb) [17.3 MB]
	Get:10 https://us.mirrors.fossho.st/armbian/apt buster/buster-desktop armhf Packages [26.6 kB]
	Get:11 https://us.mirrors.fossho.st/armbian/apt buster/buster-desktop arm64 Packages [24.7 kB]
	Get:12 https://us.mirrors.fossho.st/armbian/apt buster/buster-desktop arm64 Contents (deb) [33.4 kB]
	Get:13 https://us.mirrors.fossho.st/armbian/apt buster/buster-desktop armhf Contents (deb) [96.7 kB]
	Fetched 30.1 MB in 21s (1,452 kB/s)
	Reading package lists... Done
	
# squid logs

	1609124636.518    171 10.0.0.32 TCP_MISS/500 574 GET http://apt.armbian.com/dists/buster/InRelease - HIER_DIRECT/2605:7900:20::5 text/html
	1609125425.956    347 10.0.0.32 TCP_MISS/500 545 GET http://apt.armbian.com/dists/buster/InRelease - HIER_DIRECT/2605:7900:20::5 text/html

lanefu, 2020-12-28:
Hi.  Currently the apt.armbian.com is hard wired to redirect to https.

You'll probably want to set a specific mirror instead.  See https://docs.armbian.com/User-Guide_Fine-Tuning/#choosing-an-apt-mirror for more info

yaleman, 2020-12-28:
The is the problem though, it's returning a 500 error for some reason when using a proxy and accessing the default-configured setting in the armbian build. (`/etc/apt/sources.list.d/armbian.list:deb http://apt.armbian.com buster main buster-utils buster-desktop`)

If [this line](https://github.com/armbian/build/blob/master/lib/general.sh#L179) in /lib/general.sh has https substituted for http then that *should* fix it for everyone... I believe [apt-transport-https is part of the default build, going by this line](https://github.com/armbian/build/blob/master/lib/configuration.sh#L141).

Changing my apt config to have `https://apt.armbian.com` as the source fixed the issue, while still using the proxy.

lanefu, 2020-12-28:
Okay I'll have to look deeper into it.  Im assuming the redirect from http to https is causing issue with proxy.

This issue may solve the problem as well

https://github.com/armbian/dl-router/issues/2

igorpecovnik, 2021-03-11:
BTW. apt-cacher doesn't have such troubles.

```
curl -vx http://10.0.10.10:3142 http://apt.armbian.com/dists/buster/InRelease
*   Trying 10.0.10.10:3142...
* TCP_NODELAY set
* Connected to 10.0.10.10 (10.0.10.10) port 3142 (#0)
> GET http://apt.armbian.com/dists/buster/InRelease HTTP/1.1
> Host: apt.armbian.com
> User-Agent: curl/7.68.0
> Accept: */*
> Proxy-Connection: Keep-Alive
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Length: 18345
< Last-Modified: Tue, 09 Mar 2021 12:29:23 GMT
< Content-Type: application/octet-stream
< Date: Thu Mar 11 21:32:01 2021
< Server: Debian Apt-Cacher NG/3.3.1
< X-Original-Source: https://mirrors.netix.net/armbian/apt/dists/buster/InRelease
< Connection: Keep-Alive
``

rpardini, 2021-05-19:
> Changing my apt config to have `https://apt.armbian.com` as the source fixed the issue, while still using the proxy.

Yes, but remember that unless you have a transparent/man-in-the-middle proxy with injected certificates, https:// traffic is not cached by the proxy (it will see CONNECT method for http tunnel instead of GET).

lanefu, 2021-05-19:
Alternate solution.. you can use Armbian-config to choose a specific mirror rather than a redirect and then just edit apt sources and change the protocol..

another way to get the mirror list documented here https://docs.armbian.com/User-Guide_Fine-Tuning/#choosing-an-apt-mirror

lanefu, 2021-08-05:
redirect now persists requester protocol as well.   https://github.com/armbian/dl-router/pull/14

going to close this issue.

closed 2021-08-05 by lanefu