armbian/build #2557: Pass settings/secrets to customize-image.sh

iyesin, 2021-01-18:
Hello,

First of all, thanks everyone who develops this useful project.
I would like to apply customization to image I'm building. This customization needs extra parameters and access to secret that will be used during image customization step. This parameters and secret should not be exposed and present on image.
After reading documentation and experimenting a bit I stuck with one question: can I somehow pass additional settings/secrets to `customize-image.sh` script without calling it directly.
My question is: how can I pass settings/secret data to `customize-image.sh` script without backing it into image and hard-coding it in `customize-image.sh` script?

iyesin, 2021-01-18:
To clarify things a bit:
- During `customize-image.sh` stage I'm doing certificate generation and signing
- This creates artifacts that should be saved in image (image is per-SBC customized)
- To create artifacts I need an access to external service for which I need a secret
- This is secret is a static token
- This token should NOT be present in the resulting image

igorpecovnik, 2021-01-18:
Your files are mounted under /tmp which shouldn't go to the image.
https://github.com/armbian/build/blob/master/lib/image-helpers.sh#L109-L126

Make a test.

iyesin, 2021-01-19:
Thank you for you quick reply, @igorpecovnik!
Do you mean `userpatches/overlay/*` files?

igorpecovnik, 2021-01-19:
that folder is mounted to the image at build time.

closed 2021-01-27 by igorpecovnik