armbian/build #2558: Kernel Build failed in stable branch (v20.11) - rrl8188eu

ludiazv, 2021-01-19:
Trying to build stable branch the compilation process fails due to an error in the rtl8188eu driver.

```
drivers/net/wireless/rtl8188eu/os_dep/linux/ioctl_cfg80211.c:10276:2: error: #endif without #if
 #endif /* CONFIG_IOCTL_CFG80211 */
  ^~~~~
make[4]: *** [scripts/Makefile.build:283: drivers/net/wireless/rtl8188eu/os_dep/linux/ioctl_cfg80211.o] Error 1
```

This is new error as I've build this branch 20 days ago no errors were found.

igorpecovnik, 2021-01-19:
We don't maintain older branches very long because https://armbian.atlassian.net/browse/AR-175 sources are not fixed. Since we don't have control nor maintain 3rd party drivers, this happens easily. Its a know problem waiting for someone to start working on it. Also to fix the driver in question - its disabled on master branch for the same reason - lack of support from users side https://github.com/armbian/build/commit/c7cc1825cd9f69bebfdbe230543374da75cef566

ludiazv, 2021-01-19:
Thanks! I will disable the module also.

closed 2021-01-19 by ludiazv

igorpecovnik, 2021-01-19:
Also if you don't need any of those, build with EXTRAWIFI="no" switch.

Waxhaw, 2021-02-05:
The title above has a TYPO.  "rrl8188eu" should really be "RTL8188eu".  Caps for emphasis.