armbian/build #2560: ZFS-dkms missing dependencies

Salamandar, 2021-01-19:
I'm not 100% sure this is the place for this issue.
The zfs-dkms package is missing (at least) 2 dependencies:
* dkms
* linux-headers-current-<cpu architecture> (I don't know if it is possible).

I got this issue on the Helios64 (cpu arch = rockchip64)

EvilOlaf, 2021-01-19:
To get this straight? You have an issue building armbian with the build script and it fails at zfs? If so check `output/debug` folder for further details

Salamandar, 2021-01-19:
No, I get an error while installing the package on my armbian installation. That's why I don't know if this is the place for this kind of issues ^^''

EvilOlaf, 2021-01-19:
It is not. This issue tracker is only for issues relating to the build process.
Try here: https://forum.armbian.com/clubs/5-kobol-forum/

closed 2021-01-19 by EvilOlaf

Salamandar, 2021-02-01:
> It is not. This issue tracker is only for issues relating to the build process.
> Try here: https://forum.armbian.com/clubs/5-kobol-forum/

Please open an issue tracker for packaging issues. The forum is slow AF and not adapted to track code issues.

igorpecovnik, 2021-02-01:
> The forum is slow AF and not adapted to track code issues.

Your support contract. Read it: https://github.com/armbian/build#support

igorpecovnik, 2021-02-01:
> and not adapted to track code issues.

This is as designed. We don't track general Linux bugs by purpose, because it is totally out of range and way too big pressure to the maintainers and developers, people "you" support only with adding more work.

Maintaining and managing issues users collect without resolving them, just this task, represent a full time work for at least one person or a bunch of part time volunteers while they have no resources to appoint those tasks too. There is virtually no interest for maintenance help from you even for some simple things [we are asking you to help](https://forum.armbian.com/forum/54-help-wanted/).

We receive few euros from you per month, so we can't really hire help or pay for the costs you create. We are running the project with around 1/1000+ of needed resources ... and I hate turning you down over and over again.