armbian/build #2580: v20.11,rockpi-s, build exit with error.

kylongmu, 2021-01-27:
**Describe the bug**
Build a default config image, but exit with error.

**To Reproduce**
```
./compile.sh DOWNLOAD_MIRROR=china MAINLINE_MIRROR=tuna
choose step: Full OS->Do not change->rockpi-s->current->focal->console interface->minimal

```
**Provide logs:**

This is git log:
```
commit cce76e84f87a06fdbb1d346b11c0c10ea1d05831 (HEAD -> v20.11, origin/v20.11)
Author: Igor Pecovnik <igor.pecovnik@gmail.com>
Date:   Tue Dec 15 10:34:27 2020 +0100
```

This is build error log:
```
  CC [M]  drivers/hid/hid-sensor-hub.o
  CC [M]  drivers/hid/hid-sensor-custom.o
  LD [M]  drivers/hid/hid-logitech.o
  LD [M]  drivers/hid/hid-picolcd.o
  LD [M]  drivers/hid/hid-uclogic.o
  LD [M]  drivers/hid/wacom.o
  AR      drivers/hid/built-in.a
  LD [M]  drivers/hid/hid-wiimote.o
  LD [M]  drivers/staging/wfx/wfx.o
[ error ] ERROR in function compile_kernel [ compilation.sh:410 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
```

igorpecovnik, 2021-01-27:
> This is build error log:

**Provide logs:**
Drop logs to pastebin service of your choice and supply logs from output/debug

igorpecovnik, 2021-01-27:
Older branches of build script ( v20.11 ) we can't afford to maintain very long time and we also [have to fix this](https://armbian.atlassian.net/browse/AR-175). 

Use master branch which works:

[ o.k. ] Repeat Build Options [ ./compile.sh  BOARD=rockpi-s BRANCH=current BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=yes KERNEL_CONFIGURE=yes  ]

closed 2021-01-27 by igorpecovnik