armbian/build #2603: Building error - bullseye - odroidc4 - 21.02.0-trunk

vyn20, 2021-02-02:
./compile.sh  BOARD=odroidc4 BRANCH=dev RELEASE=bullseye BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=img

[compilation.log](https://github.com/armbian/build/files/5911498/compilation.log)

[ o.k. ] Installing base system [ Stage 2/2 ]
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
W: Failure trying to run:  /sbin/ldconfig
W: See //debootstrap/debootstrap.log for details
[ error ] ERROR in function create_rootfs_cache [ debootstrap.sh:177 ]
[ error ] Debootstrap base system second stage failed 
[ o.k. ] Process terminated 
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-dev-odroidc4-bullseye-no-yes/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:66 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated

igorpecovnik, 2021-02-02:
Sadly the problem is bug in Qemu. I already check if we can fix this somehow but this is too distant problem from our perspective. Bullseye package base is not stabilized upstream which means a few of Armbian maintainers can't do exactly nothing to change things. For possible workaround, check forum:

https://forum.armbian.com/topic/16740-debootstrap-base-system-second-stage-failed

igorpecovnik, 2021-03-02:
Workaround is to use Ubuntu 21.04 Hirsute build host.

closed 2021-03-02 by igorpecovnik

vyn20, 2021-03-03:
Perfect!