armbian/build #2649: odroidhc4 buster; won't boot with CRYPTROOT_ENABLED

Legogris, 2021-02-21:
When building armbian with the following (trying both `master`, `v21.02` and  `v2020.11` separately, from scratch) with `CRYPTROOT_ENABLE=yes`. startup gets stuck at `Starting kernel...`.

Flashing to and booting from uSD.

Works on odroidc4 (cryptroot enabled):

```
sudo ./compile.sh DEST_LANG=en_US.UTF-8 BOOTSIZE=500 BOARD=odroidc4 BRANCH=current BUILD_DESKTOP=no RELEASE=buster EXPERT=yes CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="secret!" PROGRESS_DISPLAY=plain KERNEL_ONLY=no CLEAN_LEVEL=images CRYPTROOT_SSH_UNLOCK_PORT=1234 CRYPTROOT_SSH_UNLOCK=yes
```

Works on odroidhc4 (cryptroot disabled):
```
sudo ./compile.sh DEST_LANG=en_US.UTF-8 BOOTSIZE=500 BOARD=odroidc4 BRANCH=current BUILD_DESKTOP=no RELEASE=buster EXPERT=yes PROGRESS_DISPLAY=plain KERNEL_ONLY=no CLEAN_LEVEL=images 
```

Does not work  on odroidhc4 (cryptroot enabled):
```
sudo ./compile.sh DEST_LANG=en_US.UTF-8 BOOTSIZE=500 BOARD=odroidhc4 BRANCH=current BUILD_DESKTOP=no RELEASE=buster EXPERT=yes CRYPTROOT_ENABLE=yes CRYPTROOT_PASSPHRASE="secret!" PROGRESS_DISPLAY=plain KERNEL_ONLY=no CLEAN_LEVEL=images CRYPTROOT_SSH_UNLOCK_PORT=1234 CRYPTROOT_SSH_UNLOCK=yes
```

When pressing the bottom button to force boot from uSD, it:
* finds the U-boot script at `/boot.scr`
* applies `meson-fixup.scr`
* `Executing script at 32000000`
* `Loading init Ramdisk from Legacy Image`
* flattens device tree
* Hangs after `Starting kernel...`

igorpecovnik, 2021-02-22:
This function is experimental and not supported. Perhaps prompt just doesn't show up ... Check hints in the docs: https://docs.armbian.com/Developer-Guide_Build-Options/#main-options

Legogris, 2021-02-23:
Right - noting it here because odroidc4 and odroidhc4 are almost identical apart from the SATA interfaces and the bootloader.

I did read these and experimented with a few different variations, no success yet.

Legogris, 2021-02-23:
Turned out to be a couple of things:

* Prompt just bot showing up, as you suggested. So it works over SSH, but not over keyboard (which is fine for me)
* Need to set `--type=luks1` in `CRYPTROOT_PARAMETERS`. However, the default `--pbkdf=pbkdf2` would then be overriden, and adding more  than one option breaks the script - apparently it assumes space-separated options, and the armbian build script doesn't expand that properly.

So what solved it for me was just manually setting the var `CRYPTROOT_PARAMETERS` directly in `debootstrap.sh`

closed 2021-02-23 by Legogris

igorpecovnik, 2021-02-23:
> Need to set --type=luks1 in CRYPTROOT_PARAMETERS. However, the default --pbkdf=pbkdf2 would then be overriden, and adding more than one option breaks the script - apparently it assumes space-separated options, and the armbian build script doesn't expand that properly.

If you care to fix this, you are more than welcome. There are a few people that will be glad this works (better).

Legogris, 2021-02-23:
I started attempting that but got lost trying to fond where the transition from cli arguements to variables actually happen (so just did a stupid hardcode for myself)  - if you could point me to where to look I'll take a stab at a PR :)

igorpecovnik, 2021-02-23:
https://github.com/armbian/build/blob/master/compile.sh#L253

Open a topic on forum or attach to the one related to cryptoroot.

Legogris, 2021-02-26:
This was a bit hairier to solve properly than i expected (though should be possible of course); since no other build options seem to be likely to include `=` characters, I just ended up making a note in one of the linked issues and addressing it in docs instead: https://github.com/armbian/documentation/pull/132