armbian/build #2672: armbian can't verify package signature

xlazom00, 2021-03-01:
**Describe the bug**
Can't use armbian

**To Reproduce**
pure ./compile

**Provide logs:**
```
[ o.k. ] Checking for external GCC compilers 
[ error ] ERROR in function download_and_verify [ general.sh:1382 ]
[ error ] verification failed 
[ o.k. ] Process terminated 

fron logs
Displaying message: Checking for external GCC compilers  info
pub   rsa2048 2011-01-04 [SC]
      873D6A08B28A5167B466D9D4D47877808F427EAF
uid           [ unknown] Linaro Toolchain Builder <michael.hope+cbuild@linaro.org>

pub   rsa4096 2015-03-16 [SC]
      DF00FAF1C577104B50BF1D0093D6889F9F0E78D5
uid           [ unknown] Igor Pecovnik <igor@armbian.com>
uid           [ unknown] Igor Pecovnik (Ljubljana, Slovenia) <igor.pecovnik@gmail.com>
sub   rsa4096 2015-03-16 [E]

gpg: assuming signed data in '/home/misko/build_org/build/cache/toolchain/gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz'
gpg: Signature made Tue 19 Nov 2013 01:00:03 AM PST
gpg:                using RSA key D47877808F427EAF
gpg: BAD signature from "Linaro Toolchain Builder <michael.hope+cbuild@linaro.org>" [unknown]
Displaying message: ERROR in function download_and_verify general.sh:1382 err
Displaying message: verification failed  err
Displaying message: Process terminated  info
```

igorpecovnik, 2021-03-01:
Did you try again? Or its always like this?

xlazom00, 2021-03-01:
10 times with same output

igorpecovnik, 2021-03-01:
Try removing cache?

xlazom00, 2021-03-02:
armbian cache ?
git clean -dfx
git reset --hard
and still this
```[ .... ] downloading from mirrors.netix.net using http(s) network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]

[ error ] ERROR in function download_and_verify [ general.sh:1382 ]
[ error ] verification failed ```

igorpecovnik, 2021-03-02:
cache folder inside build/

xlazom00, 2021-03-02:
still same :(
```
[ o.k. ] Checking for external GCC compilers                                                                                                                                                                                                                                 │  
[ .... ] downloading from mirrors.netix.net using http(s) network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]                                                                                                                                                   │  
  │                                                                                                                                                                                                                                                                          │  
[ error ] ERROR in function download_and_verify [ general.sh:1382 ]                                                                                                                                                                                                          │  
[ error ] verification failed```

xlazom00, 2021-03-02:
And I am testing ubuntu 21.04(just installed)
with same results :(
[ o.k. ] Checking for external GCC compilers 
[ error ] ERROR in function download_and_verify [ general.sh:1382 ]
[ error ] verification failed 
[ o.k. ] Process terminated

xlazom00, 2021-03-02:
Or so file gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz
looks like some html
```
cat gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<html>
 <head>
  <title>Index of /armbian/dl</title>
 </head>
 <body>
<h1>Index of /armbian/dl</h1>
  <table>
   <tr><th valign="top"><img src="/icons/blank.gif" alt="[ICO]"></th><th><a href="?C=N;O=D">Name</a></th><th><a href="?C=M;O=A">Last modified</a></th><th><a href="?C=S;O=A">Size</a></th><th><a href="?C=D;O=A">Description</a></th></tr>
   <tr><th colspan="5"><hr></th></tr>
<tr><td valign="top"><img src="/icons/back.gif" alt="[PARENTDIR]"></td><td><a href="/armbian/">Parent Directory</a></td><td>&nbsp;</td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="_rootfs/">_rootfs/</a></td><td align="right">2021-03-01 16:04  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="_toolchain/">_toolchain/</a></td><td align="right">2020-11-23 00:43  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapi/">bananapi/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapim2plus/">bananapim2plus/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapim2ultra/">bananapim2ultra/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapim2zero/">bananapim2zero/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapim3/">bananapim3/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapim64/">bananapim64/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapipro/">bananapipro/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR]"></td><td><a href="bananapir2/">bananapir2/</a></td><td align="right">2021-03-01 16:54  </td><td align="right">  - </td><td>&nbsp;</td></tr>
<tr><td valign="top"><img src="/icons/folder.gif" alt="[DIR```

xlazom00, 2021-03-02:
ok so the problem is that armbian will download from
https://minio.k-space.ee/armbian/dl//_toolchain/gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz
and double slash is problem :(

closed 2021-03-02 by igorpecovnik