armbian/build #2673: Plz add ubuntu 20.10 back

xlazom00, 2021-03-01:
**Describe the bug**
ubuntu 20.10 is missing

**To Reproduce**
run ./compile

**Provide logs:**
When I add build/config/distributions/groovy still I get this error
```
[ o.k. ] ... remote not found [ Creating new rootfs cache for groovy ]
[ o.k. ] Installing base system [ Stage 1/2 ]
E: unrecognized or invalid option --include=
[ error ] ERROR in function create_rootfs_cache [ debootstrap.sh:177 ]
[ error ] Debootstrap base system for dev odroidc4 groovy   no first stage failed 
[ o.k. ] Process terminated 
[ o.k. ] Unmounting [ /home/misko/build_org/build/.tmp/rootfs-6ccac526-a43c-465f-a2dd-4a0a20246f30/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:65 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

igorpecovnik, 2021-03-01:
Those mid-releases are usually go out pretty quick. But we provide 21.04 / Hirsute instead. Remember that you can only build it on Hirsute host ...

xlazom00, 2021-03-01:
@igorpecovnik but 21.04 isn't out, so can we get 20.10 back?  20.10 and 21.04 will have same version of gnome so when I want to test panfrost 20.10 is good enought :)

igorpecovnik, 2021-03-01:
> so can we get 20.10 back?

You can try adding it on your own. With recent major refactoring, it shouldn't be hard.

> but 21.04 isn't out

It's out for testing - as within Armbian. For testing / playing around. 

> Do you have any idea where is the problem in

I answered to that question before you asked :) "Remember that you can only **build it on Hirsute host** ..." Building on Focal is failing.

xlazom00, 2021-03-02:
@igorpecovnik but we was able to build 20.10 on 20.04 so can we also revert that piece of code?

igorpecovnik, 2021-03-02:
> can we also revert that piece of code?

Not simple anymore after this https://github.com/armbian/build/pull/2662 I already lost up to 10h per day for maintaining this project ...

closed 2021-03-02 by xlazom00