armbian/build #2692: armbian-hardware-optimization breaks log rotating when ramlog is enabled

NekoB0x, 2021-03-06:
https://github.com/armbian/build/blob/520f51de972047c5aa35b9d14669ae15895ab8ad/packages/bsp/common/usr/lib/armbian/armbian-hardware-optimization#L54

On Armbian 21.02.2 this script looks for "/dev/zram0" but if the ramlog uses "/dev/zram1" it edits all the configs in /etc/logrotate.d back to "/var/log", this breaks log rotating when ramlog is enabled.

Steps to reproduce the behavior:

Ramlog at /dev/zram1 (ramlog is enabled, zram swap not disabled, the defaults)

NAME       ALGORITHM DISKSIZE  DATA  COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram1                50M  2.9M 590.4K 1012K       6 /var/log
/dev/zram0               1.9G 55.4M  23.1M   24M       6 [SWAP]

I think a probable fix would be:
`if [[ "${CheckDevice}" == *"/dev/zram"* || "${CheckDevice}" == "armbian-ramlog" ]]; then`

igorpecovnik, 2021-03-11:
Thanks!

> I think a probable fix would be:

It looks that way. Can you do a quick test and make a pull request?

NekoB0x, 2021-03-12:
@igorpecovnik Done, still new to this, hope I didn't mess up somewhere.

igorpecovnik, 2021-03-13:
Looks alright, thanks!

PR #2705 (2021-03-13): Fix armbian-hardware-optimization logrotate editing
# Description

Fixes #2692, editing of logrotate configs when ramlog is enabled and /var/log is mounted on zram device other than /dev/zram0.

# How Has This Been Tested?

Please describe the tests that you ran to verify your changes. Please also note any relevant details for your test configuration.

- [x] Tested on a live system with reboot

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [  [... cut]
closed 2021-03-13 by igorpecovnik