armbian/build #2712: nls_ascii is built as module instead of built-in

maz-1, 2021-03-14:
When root filesystem fails and drops to initramfs, it is impossible to mount fat32 boot partition to correct and error
the error message is "IO charset ascii not found".
I believe nls_ascii should be built-in to solve this problem

igorpecovnik, 2021-03-24:
You can try to send PR by adding misins config option to those files https://github.com/armbian/build/tree/master/config/kernel or wait we do it. Just that can take a while.

AristoChen, 2021-05-22:
Hi, 

Looks like this problem is still not fixed yet, so I would like to give it a try.

from my understanding, the field called `CONFIG_NLS_ASCII`  in all the [config files](https://github.com/armbian/build/tree/master/config/kernel) should be like `CONFIG_NLS_ASCII=y`.

and here is what I found
```shell
linux-imx6-current.config:CONFIG_NLS_ASCII=m
linux-imx6-edge.config:CONFIG_NLS_ASCII=m
linux-imx7d-legacy.config:CONFIG_NLS_ASCII=y
linux-jetson-nano-current.config:CONFIG_NLS_ASCII=m
linux-jetson-nano-edge.config:CONFIG_NLS_ASCII=m
linux-meson-current.config:CONFIG_NLS_ASCII=m
linux-meson-edge.config:CONFIG_NLS_ASCII=m
linux-meson-legacy.config:CONFIG_NLS_ASCII=m
linux-meson64-current.config:CONFIG_NLS_ASCII=m
linux-meson64-edge.config:CONFIG_NLS_ASCII=m
linux-meson64-legacy.config:CONFIG_NLS_ASCII=m
linux-mt7623-edge.config:# CONFIG_NLS_ASCII is not set
linux-mt7623-legacy.config:# CONFIG_NLS_ASCII is not set
linux-mvebu-current.config:CONFIG_NLS_ASCII=m
linux-mvebu-edge.config:CONFIG_NLS_ASCII=m
linux-mvebu-legacy.config:# CONFIG_NLS_ASCII is not set
linux-mvebu64-current.config:CONFIG_NLS_ASCII=m
linux-mvebu64-edge.config:CONFIG_NLS_ASCII=m
linux-mvebu64-legacy.config:# CONFIG_NLS_ASCII is not set
linux-odroidxu4-current.config:CONFIG_NLS_ASCII=m
linux-odroidxu4-edge.config:CONFIG_NLS_ASCII=y
linux-odroidxu4-legacy.config:CONFIG_NLS_ASCII=y
linux-rk322x-current.config:CONFIG_NLS_ASCII=y
linux-rk322x-edge.config:CONFIG_NLS_ASCII=y
linux-rk322x-legacy.config:CONFIG_NLS_ASCII=y
linux-rk3399-legacy.config:CONFIG_NLS_ASCII=m
linux-rockchip-current.config:CONFIG_NLS_ASCII=y
linux-rockchip-edge.config:CONFIG_NLS_ASCII=y
linux-rockchip-legacy.config:CONFIG_NLS_ASCII=y
linux-rockchip64-current.config:CONFIG_NLS_ASCII=m
linux-rockchip64-edge.config:CONFIG_NLS_ASCII=m
linux-rockchip64-legacy.config:CONFIG_NLS_ASCII=y
linux-rockpis-edge.config:# CONFIG_NLS_ASCII is not set
linux-rockpis-legacy.config:CONFIG_NLS_ASCII=y
linux-s5p6818-legacy.config:CONFIG_NLS_ASCII=m
linux-station-legacy.config:CONFIG_NLS_ASCII=y
linux-sunxi-current.config:CONFIG_NLS_ASCII=m
linux-sunxi-edge.config:CONFIG_NLS_ASCII=m
linux-sunxi-legacy.config:CONFIG_NLS_ASCII=m
linux-sunxi64-current.config:CONFIG_NLS_ASCII=m
linux-sunxi64-edge.config:CONFIG_NLS_ASCII=m
linux-sunxi64-legacy.config:CONFIG_NLS_ASCII=m
linux-virtual-current.config:CONFIG_NLS_ASCII=m
linux-zynq-legacy.config:CONFIG_NLS_ASCII=m
```

so I guess the solution is to make `CONFIG_NLS_ASCII=m` and `# CONFIG_NLS_ASCII is not set` become `CONFIG_NLS_ASCII=y`? 

please correct me if I misunderstood anything. thanks!

igorpecovnik, 2021-05-22:
> so I guess the solution is to make CONFIG_NLS_ASCII=m and # CONFIG_NLS_ASCII is not set become CONFIG_NLS_ASCII=y?

Yes.

closed 2021-07-23 by maz-1