armbian/build #2730: Error when building image for rockpi-e

fuji246, 2021-03-19:
Any thoughts?

```
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ focal ]
[ o.k. ] Syncing clock [ host ]
[ o.k. ] Checking for external GCC compilers 
[ o.k. ] Downloading sources 
[ o.k. ] Checking git sources [ u-boot v2020.10 ]
warning: redirecting to https://source.denx.de/u-boot/u-boot.git/
warning: redirecting to https://source.denx.de/u-boot/u-boot.git/
[ .... ] Up to date 
[ o.k. ] Checking git sources [ linux-mainline linux-5.10.y ]
[ .... ]  Cleaning ....  [ 222 files ]
[ o.k. ] Checking git sources [ sunxi-tools master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ rkbin-tools master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ marvell-tools A3700_utils-armada-18.12 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ marvell-ddr mv_ddr-armada-18.12 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ marvell-binaries binaries-marvell-armada-18.12 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ odroidc2-blobs master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ testing-reports master ]

[ .... ] Up to date 
[ o.k. ] Checking git sources [ amlogic-boot-fip master ]
[ .... ] Up to date 
[ o.k. ] Adjusting [ packaging ]
[ o.k. ] * applying general-packaging-5.10.y.patch 
[ o.k. ] Adding [ Kernel splash file ]
[ o.k. ] * applying bootsplash-5.8.10-0001-Revert-vgacon-remove-software-scrollback-support.patch 
[ o.k. ] * applying bootsplash-5.8.10-0002-Revert-fbcon-remove-now-unusued-softback_lines-curso.patch 
[ o.k. ] * applying bootsplash-5.10.y-0003-Revert-fbcon-remove-soft-scrollback-code.patch 
[ o.k. ] * applying 0001-bootsplash.patch 
[ o.k. ] * applying 0002-bootsplash.patch 
[ o.k. ] * applying 0003-bootsplash.patch 
[ o.k. ] * applying 0004-bootsplash.patch 
[ o.k. ] * applying 0005-bootsplash.patch 
[ o.k. ] * applying 0006-bootsplash.patch 
[ o.k. ] * applying 0007-bootsplash.patch 
[ o.k. ] * applying 0008-bootsplash.patch 
[ o.k. ] * applying 0009-bootsplash.patch 
[ o.k. ] * applying 0010-bootsplash.patch 
[ o.k. ] * applying 0011-bootsplash.patch 
[ o.k. ] * applying 0012-bootsplash.patch 
[ o.k. ] Adding [ Wireless package injections for mac80211 compatible chipsets ]
[ o.k. ] * applying kali-wifi-injection-1-v5.9-post.patch 
[ o.k. ] * applying kali-wifi-injection-2.patch 
[ o.k. ] * applying kali-wifi-injection-3.patch 
[ o.k. ] Adding [ AUFS 5.10 ]
[ o.k. ] Checking git sources [ aufs5 aufs5.10 ]
[ .... ] Up to date 
[ o.k. ] * applying aufs5-kbuild.patch 
[ o.k. ] * applying aufs5-base.patch 
[ o.k. ] * applying aufs5-mmap.patch 
[ o.k. ] * applying aufs5-standalone.patch 
[ o.k. ] Adding [ Drivers for 2.5Gb RTL8152/RTL8153 USB dongles branch:master ]
[ o.k. ] Checking git sources [ rtl8152 master ]
[ .... ] Up to date 
[ o.k. ] Adding [ Wireless drivers for Realtek 8189ES chipsets branch:master ]
[ o.k. ] Checking git sources [ rtl8189es master ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8189FS chipsets branch:rtl8189fs ]
[ o.k. ] Checking git sources [ rtl8189fs rtl8189fs ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Adding [ Wireless drivers for Realtek 8192EU chipsets branch:realtek-4.4.x ]
[... 110 lines cut ...]
[ o.k. ] * [l][c] rk3328-dtsi-mmc-vdec-usb3-tweaks.patch 
[ o.k. ] * [l][c] rk3328-fix-mali-node.patch 
[ o.k. ] * [l][c] rk3399-add-sclk-i2sout-src-clock.patch 
[ o.k. ] * [l][c] rk3399-bump-voltages-for-low-opps.patch 
[ o.k. ] * [l][c] rk3399-enable-dwc3-xhci-usb-trb-quirk.patch 
[ warn ] * [l][c] rk3399-pci-rockchip-support-ep-gpio-undefined-case.patch [ failed ]
[ o.k. ] * [l][c] rk3399-sd-drive-level-8ma.patch 
[ o.k. ] * [l][c] rk3399-unlock-temperature.patch 
[ o.k. ] * [l][c] wifi-4003-fix-sha256_state-clashes.patch 
[ o.k. ] * [l][c] wifi-4004-fix-cfg80211-for-5.8.patch 
[ .... ] Compressing sources for the linux-source package 
1.77GiB [10.1MiB/s] [===================================================================================================================================================================================================================================] 102%
[ o.k. ] Compiling current kernel [ 5.10.24 ]
[ o.k. ] Compiler version [ aarch64-linux-gnu-gcc 8.3.0 ]
[ o.k. ] Using kernel config file [ config/kernel/linux-rockchip64-current.config ]
#
# configuration written to .config
#

...

[ error ] ERROR in function compile_kernel [ compilation.sh:413 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
```

EvilOlaf, 2021-03-19:
Check `output/debug` folder for the cause.

EvilOlaf, 2021-03-19:
```
[ warn ] * [l][c] general-add-dwc3-xhci-usb-trb-quirk.patch [ failed ]

[ warn ] * [l][c] general-bluetooth-add-new-quirk.patch [ failed ]

[ warn ] * [l][c] rk3399-pci-rockchip-support-ep-gpio-undefined-case.patch [ failed ]
```
Maybe stuff has been upstreamed?

piter75, 2021-03-19:
@fuji246 Make sure you are using the latest master
Those patches were adjusted a few days ago to account for the kernel changes: https://github.com/armbian/build/commit/a3d29f40826ade6a5e2dea214bac5b4c08de8ea8

fuji246, 2021-03-19:
I've updated to master right after it failed the first time.

In `output/debug/compilation.log`:

```

        == kernel ==

scripts/Makefile.lib:8: 'always' is deprecated. Please use 'always-y' instead
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:22.11-27.6: Warning (spi_bus_reg): /fragment@1/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:36.11-41.6: Warning (spi_bus_reg): /fragment@2/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:50.11-55.6: Warning (spi_bus_reg): /fragment@3/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:64.11-69.6: Warning (spi_bus_reg): /fragment@4/__overlay__/spidev: SPI bus unit address format error, expected "0"
drivers/usb/host/xhci-ring.c: In function ‘xhci_queue_bulk_tx’:
drivers/usb/host/xhci-ring.c:3414:23: error: ‘XHCI_TRB_ENT_QUIRK’ undeclared (first use in this function); did you mean ‘XHCI_EP_LIMIT_QUIRK’?
    if (xhci->quirks & XHCI_TRB_ENT_QUIRK && en_trb_ent)
                       ^~~~~~~~~~~~~~~~~~
                       XHCI_EP_LIMIT_QUIRK
drivers/usb/host/xhci-ring.c:3414:23: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [scripts/Makefile.build:279: drivers/usb/host/xhci-ring.o] Error 1
make[2]: *** [scripts/Makefile.build:496: drivers/usb/host] Error 2
make[1]: *** [scripts/Makefile.build:496: drivers/usb] Error 2
make: *** [Makefile:1807: drivers] Error 2
make: *** Waiting for unfinished jobs....
```

fuji246, 2021-03-19:
And in patching log:

```
Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/board-rockpro64-fix-emmc.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/board-rockpro64-fix-spi1-flash-speed.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/board-rockpro64-work-led-heartbeat.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-add-dwc3-xhci-usb-trb-quirk.patch
1 out of 2 hunks FAILED -- saving rejects to file drivers/usb/host/xhci.h.rej

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-add-overlay-compilation-support.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-add-overlay-configfs.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-add-xtx-spi-nor-chips.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-bluetooth-add-new-quirk.patch
1 out of 1 hunk FAILED -- saving rejects to file include/net/bluetooth/hci.h.rej

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-btsdio-ignore-uart-devs.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-disable-mtu-validation.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-emmc-hs400es-init-tweak.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-fix-es8316-kernel-panic.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-fix-mmc-signal-voltage-before-reboot.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-increasing_DMA_block_memory_allocation_to_2048.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-possibility-of-disabling-rk808-rtc.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-rkvdec_nv15.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-rockchip-overlays.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-rt5651-add-mclk.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/overlays-00-add-oc-opp-rk3399.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/overlays-01-add-oc-opp-rk3328.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3328-add-dmc-driver.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3328-audio-and-renegade-supplies.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3328-dtsi-mmc-vdec-usb3-tweaks.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3328-fix-mali-node.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3399-add-sclk-i2sout-src-clock.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3399-bump-voltages-for-low-opps.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3399-enable-dwc3-xhci-usb-trb-quirk.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3399-pci-rockchip-support-ep-gpio-undefined-case.patch
1 out of 1 hunk FAILED -- saving rejects to file drivers/pci/controller/pcie-rockchip.c.rej

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3399-sd-drive-level-8ma.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/rk3399-unlock-temperature.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/wifi-4003-fix-sha256_state-clashes.patch

Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/wifi-4004-fix-cfg80211-for-5.8.patch
```

piter75, 2021-03-19:
> I've updated to master right after it failed the first time

This line:
```
Processing file /home/vagrant/armbian/patch/kernel/rockchip64-current/general-add-dwc3-xhci-usb-trb-quirk.patch
1 out of 2 hunks FAILED -- saving rejects to file drivers/usb/host/xhci.h.rej
```
convinces me that somehow your build is not using the latest master as the file was renamed/disabled 2 days ago by Igor: https://github.com/armbian/build/commit/3c96df8d2577176cb483591f1a9092b5e90901c8

Please post the result of the command: `git rev-parse HEAD` and the command line that you are using to build the image.

fuji246, 2021-03-20:
```
vagrant@ubuntu-focal:~/armbian$ git rev-parse HEAD
bf544c21f8778e0b992b835baae9c6ed6b2e2075
```

EvilOlaf, 2021-03-20:
Your git seems to be behind: https://github.com/armbian/build/commit/bf544c21f8778e0b992b835baae9c6ed6b2e2075

If git pull does not do the trick try to start from scratch and clone the whole repo again.

fuji246, 2021-03-21:
forked a new branch seems fixed the issue, thanks a lot! However as I showed above the commit is `bf544c21f8778e0b992b835baae9c6ed6b2e2075` which should be the latest, any thoughts on how to clean the build?

piter75, 2021-03-21:
> However as I showed above the commit is `bf544c21f8778e0b992b835baae9c6ed6b2e2075` which should be the latest

As @EvilOlaf pointed out in his previous reply it is not the latest commit.

> forked a new branch seems fixed the issue, thanks a lot!

If you are building with your own branch then you need to rebase it on master or merge from master from time to time.

If you are building with the forked repository then you should sync it with the upstream from time to time.
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/syncing-a-fork

Only this way you can get the latest changes.

Closing as it is not a build system issue.

closed 2021-03-21 by piter75