armbian/build #2734: Errors building out-of-tree modules on OdroidC2 using Armbian version 21.02.3

Lamarqe, 2021-03-21:
Hello, 

This is a highly similar issues as reported [here](https://forum.armbian.com/topic/16912-nanopi-neo-2-armbian-201110-focal-unable-to-build-kernel-5104-source-no-rule-to-make-target-archarm64kernelvdsovdsolds/?tab=comments#comment-118296) and resolved [here](https://armbian.atlassian.net/browse/AR-608) for the NanoPI.

I see that the fix is not NanoPI-specific, however I still run into the same problem, using the most recent Armbian Kernel (version  21.02.3, kernel 5.10.21):

```
root@odroidc2:/usr/src/linux-headers-5.10.21-meson64# make KERNELVERSION=5.10.21-meson64 -j3 modules_prepare
scripts/Makefile.build:414: warning: overriding recipe for target 'modules.order'
Makefile:1401: warning: ignoring old recipe for target 'modules.order'
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vdso.lds', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.  Stop.
make: *** [arch/arm64/Makefile:186: vdso_prepare] Error 2
```

Am I doing something wrong or is this issue not yet fully resolved?

igorpecovnik, 2021-03-24:
This is probably something new from upstream. Needs to be investigated.

Lamarqe, 2021-03-27:
Ok, thanks for confirming. I would be willing to invest time and look into it. However, I have no idea where and how to start.

igorpecovnik, 2021-05-22:
Can you try to recreate this on latest image - i think we nailed this down.

Lamarqe, 2021-05-22:
Thank you for following up on this item! I spent a significant amount of my time on this problem last weekend by building the kernel myself an comparing this with the makefiles of the headers only. But I could not get anywhere with this. :(

I tried it once more using this build:
https://beta.armbian.com/pool/main/l/linux-5.11.22-meson64/linux-headers-edge-meson64_21.08.0-trunk.27_arm64.deb

However, the build error is still unchanged:
`scripts/Makefile.build:414: warning: overriding recipe for target 'modules.order'
Makefile:1420: warning: ignoring old recipe for target 'modules.order'
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vdso.lds', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.  Stop.
make: *** [arch/arm64/Makefile:191: vdso_prepare] Error 2
`

igorpecovnik, 2021-05-22:
Ahaa, I see. Nobody made a proper fix after my unsuccessful attempt. The core of the problem is that this file is missing. Adjustment of Makefile or some dirty hack is needed. Strange is that its not fixed elsewhere too.

Lamarqe, 2021-05-22:
Yes, you are right. 
The file scripts/modules.lds is gone again. 

When I reported the problem in March, using kernel 5.10.21, the file was present (see my initial report above)

igorpecovnik, 2021-05-22:
We merged changed packaging and kernel 5.12.y ... worth trying.

Lamarqe, 2021-05-22:
Is this package already somewhere available or shall I simply wait until it appears at https://beta.armbian.com/pool/main/l/ ?

igorpecovnik, 2021-05-22:
They are already there.

Lamarqe, 2021-05-22:
It is still unchanged, also with todays build of kernel 5.12
```
scritch@odroidc2:/usr/src/linux-headers-5.12.6-meson64$ sudo make KERNELVERSION=5.12.6-meson64 -j3 modules_prepare
scripts/Makefile.build:421: warning: overriding recipe for target 'modules.order'
Makefile:1451: warning: ignoring old recipe for target 'modules.order'
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vdso.lds', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.  Stop.
make: *** [arch/arm64/Makefile:191: vdso_prepare] Error 2
```

igorpecovnik, 2021-05-22:
OK, then bug wasn't fixed upstream neither. I don't have time to deal with this once again anytime soon.

Lamarqe, 2021-05-22:
Ok, sad to hear. :(

Just as an fyi:

With the new 5.12 build, the file scripts/module.lds disappeared again.
I copied it over manually, but that does not show any effect.

igorpecovnik, 2021-05-22:
> Just as an fyi:

Just remember that those bugs are your cost:
https://github.com/armbian/build#support

Lamarqe, 2021-05-22:
Thank you for the reminder. This is understood.

As indicated above, I tried and might try again to look into this myself. 
Your armbian kernel build scripts are awesome. 

However, analyzing the kernel build and its makefiles appears to be complex and not an easy task (at least for me).

Lamarqe, 2021-10-21:
Wow, it is great to see that this issue has been resolved! 
Thank you so much.

I just retried to build additional modules, but I don't even reach the point of the previous error anymore.
The build still fails in the modules_prepare step, but in an earlier phase:

```
root@odroidc2:/usr/src/linux-headers-5.14.11-meson64# sudo make KERNELVERSION=5.14.11-meson64 -j3 modules_prepare
make[1]: *** No rule to make target 'arch/arm64/tools/gen-cpucaps.awk', needed by 'arch/arm64/include/generated/asm/cpucaps.h'.  Stop.
make: *** [arch/arm64/Makefile:170: archprepare] Error 2
```

Any ideas on this one?

Geremia, 2022-03-15:
@Lamarqe I this same issue with 5.10.103-v8+. Running your `wget`s and `chmod +x /usr/src/linux-headers-5.10.103-v8+/arch/arm64/kernel/vdso/gen_vdso_offsets.sh` fixed some things, but now I get:
```
No rule to make target 'init/main.o', needed by 'init/built-in.a'.  Stop.
```

Lamarqe, 2022-03-15:
Sorry, I have no idea how to fix this. 

I was able to resolve the compile errors back then but had to figure out right afterwards that what I actually tried to accomplish was not possible with the kernel module which I was then able to compile. 
That means that there is currently no use-case for me anymore and I don't even have this setup anymore. 
Sorry.

gegtor, 2022-04-03:
Same here

Headers and source installed via armbian config

`No rule to make target 'arch/arm64/kernel/vdso/vdso.lds'`

JM2707, 2022-04-10:
Same problem here. Edge kernel does not fix it.

hqnicolas, 2024-06-21:
Same problem here. Edge kernel does not fix it.
https://github.com/hqnicolas/linux_openvfd/blob/master/SETUP.md