armbian/build #2739: "armbian-ramlog write" is called twice

NekoB0x, 2021-03-23:
**Describe the bug**

armbian-ramlog write gets called in a cron job and in logrotate.service, both at the same time.

https://github.com/armbian/build/blob/5e69351b3dc05829133e893f6efe575ef58c7866/packages/bsp/common/etc/cron.daily/armbian-ram-logging#L2

https://github.com/armbian/build/blob/5e69351b3dc05829133e893f6efe575ef58c7866/packages/bsp/common/etc/systemd/system/logrotate.service#L8

I suggest removing the cron job.

igorpecovnik, 2021-03-24:
Hmm, in theory those are two different services and in both cases write should be conducted. Do we experience troubles because of this?

NekoB0x, 2021-03-24:
@igorpecovnik 
On new distros (Focal) with logrotate.service this script is called twice exactly at 0:00 each day, at worse it probably can cause the logs to be written twice (more flash wear) or some race conditions between two rsync processes but I didn't test this thoroughly.
On old distros (Bionic) there is only logrotate.service which is included with armbian root, but no logrotate.timer therefore it does nothing and only the cron job executes (old partially broken behavior with logrotate being a cron job).

An easy fix would be adding a check for logrotate.timer.

PR #2741 (2021-03-25): Fix redundant log writing
# Description

Fixes #2739, calling "armbian-ramlog write" twice on systems with logrotate systemd unit.

# How Has This Been Tested?

- [x] Running the script manually.

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] My changes generate no new warnings
closed 2021-03-25 by lanefu