armbian/build #2742: Hummingboard 2 support

joker5bb, 2021-03-25:
On the website Hummingboard 2 is stated that is supported. 
https://www.armbian.com/hummingboard-2/

I downloaded the latest image from link above and it does boot using the Cubox-i image.

Cubox-i is a separate hardware with similar features. 
https://www.armbian.com/cubox-i/

Armbian is recommending to use Cubox-i image for Hummingboard 2 
They both use the Freescale i.MX6 SoC 
The problem is the mpcie and emmc are not working for Hummingboard 2 

I need to custom build Armbian to correct the kernel config but I have to choose Cubox-i as the board 
The compile fails with no modifications 
Displaying message: Cleaning u-boot/v2018.01-solidrun-imx6 info
Displaying message: Compiling u-boot 2018.01 info
Displaying message: Compiler version arm-none-linux-gnueabihf-gcc 9.2.1 info
Displaying message: Checking out to clean sources
Displaying message: Cleaning u-boot/v2018.01-solidrun-imx6 info
Displaying message: Started patching process for u-boot imx6-cubox-i-current info
Displaying message: Looking for user patches in userpatches/u-boot/u-boot-imx6 info
Displaying message: * [\e[32ml\e[0m][\e[35mb\e[0m] change_logo_to_armbian.patch  info
Displaying message: ERROR in function compile_uboot compilation.sh:235 err
Displaying message: U-boot compilation failed  err
Displaying message: Process terminated  info

EvilOlaf, 2021-03-25:
Cannot reproduce. U-Boot/kernel packages building flawless for me.
Check `output/debug` folder for more detailed logs.

igorpecovnik, 2021-03-25:
Solidrun U-boot, which we use, auto detects hw which is why we only have one board configuration. But since there are many revisions and since its not very well supported, sometimes misses. It might help if you force respective DT blob for hummingboard2 in armbianEnv.txt, add fdtfile=hummingboard2.dtb

joker5bb, 2021-03-26:
I was a able to compile by switching from native to virtual Ubuntu Focal 20.04.2 setup

Freescale i.MX6/7/8 PCIe controller
modulename: pci-imx6.ko
configname: CONFIG_PCI_IMX6 

the controller is enabled by default, but kernel 5.11.10 did not work for me.
will try  kernel 5.12

igorpecovnik, 2021-03-26:
Not a build script issues. Closing.

closed 2021-03-26 by igorpecovnik