armbian/build #2757: The system hangs close to the build end.

starkovv, 2021-04-08:
### Issue description:
I have build environment inside LXD container on my host.

The system hangs close to the build end.

### Steps to reproduce the behavior:

**Step #1. On host. Launch LXD container and prepare build environment:**
```
lxc launch ubuntu:20.04 armbian-builder -c security.nesting=true

lxc exec armbian-builder bash

apt-get update && apt-get upgrade -y

apt install -y apt-transport-https ca-certificates curl software-properties-common

curl -fsSL https://download.docker.com/linux/ubuntu/gpg | apt-key add -

add-apt-repository "deb [arch=amd64] https://download.docker.com/linux/ubuntu bionic stable"

apt update

apt-cache policy docker-ce

apt install -y docker-ce

systemctl status docker
```

**Step #2. Inside LXD container. Run build process.**
```
apt-get -y -qq install git  
git clone --depth 1 https://github.com/armbian/build  
cd build 
./compile.sh BRANCH=edge
```

### Logs:
```
[...]
remote: Enumerating objects: 271, done.
remote: Counting objects: 100% (271/271), done.
remote: Compressing objects: 100% (217/217), done.
remote: Total 271 (delta 40), reused 260 (delta 39), pack-reused 0
Receiving objects: 100% (271/271), 10.91 MiB | 2.58 MiB/s, done.
Resolving deltas: 100% (40/40), done.
From https://github.com/armbian/firmware
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Checking git sources [ armbian-firmware-git master ]
[ .... ] Up to date
[ o.k. ] Checking git sources [ linux-firmware-git master ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Enumerating objects: 2376, done.
remote: Counting objects: 100% (2376/2376), done.
remote: Compressing objects: 100% (1666/1666), done.
remote: Total 2376 (delta 857), reused 1919 (delta 646), pack-reused 0
Receiving objects: 100% (2376/2376), 219.36 MiB | 2.80 MiB/s, done.
Resolving deltas: 100% (857/857), done.
From git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware
 * branch            master     -> FETCH_HEAD
 * [new branch]      master     -> origin/master
[ .... ] Checking out
[ o.k. ] Creating board support package [ nanopi-r4s edge ]
[ .... ] Fingerprinting
[ o.k. ] Building package [ linux-focal-root-edge-nanopi-r4s ]
[ o.k. ] Starting rootfs and image building process for [ edge nanopi-r4s focal   no ]
[ o.k. ] Checking for local cache [ focal-cli-arm64.e1e...b15.tar.lz4 ]
[ .... ] searching on servers
[ .... ] downloading using torrent network [ focal-cli-arm64.e1e6bb3f60f87aea66dc527523737b15.tar.lz4 ]
[ o.k. ] Verified [ PGP ]
[ o.k. ] Extracting focal-cli-arm64.e1e...b15.tar.lz4 [ 0 days old ]
[ .... ] focal-cli-arm64.e1e...b15.tar.lz4:  350MiB [ 255MiB/s] [=========================================================================>] 100%
[ o.k. ] Adding Armbian repository and authentication key [ /etc/apt/sources.list.d/armbian.list ]
mount: /root/r4s/build/.tmp/rootfs-4e584a77-1c87-4422-b061-c268b5dc371b/dev: permission denied.
[ o.k. ] Applying distribution specific tweaks for [ focal ]
[ o.k. ] Applying common tweaks
[ .... ] Updating [ package lists ]
```
At this point one of CPU on host eats 100% and nothing happen.

starkovv, 2021-04-08:
UPDATE:

The build hangs at process `qemu-aarch64-static /usr/lib/apt/methods/http`.

Here is strace:
```
Trace of process 2103118 - /usr/bin/qemu-aarch64-static /usr/lib/apt/methods/http                           
trace: Process 2103118 attached                                                                            
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999997935})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999998552})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999998603})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999157})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999230})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999227})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999216})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999217})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999220})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999222})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999216})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999216})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999222})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999234})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999211})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999220})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999217})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999998641})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999201})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999998993})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999182})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999012})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999057})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999999202})
[... cut ...]
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999998460})
ead(3, "", 64965)                      = 0                                                                 
select6(4, [0 3], [], NULL, {tv_sec=30, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=29, tv_nsec=999998854})
ead(3, "", 64965)                      = 0                                                                 
```

igorpecovnik, 2021-04-08:
> qemu-aarch64-static

It seems you encounter yet another (corner case) bug related to Qemu where we can't do nothing. File a bug to Qemu or try with some other type of virtualization. I use KVM where things works - compiling Focal image works on Focal host, while Qemu crashes if you want to build Hirsute or Bullseye target. For that, you need to use Hirsute host. That is also one option to follow up.

starkovv, 2021-04-08:
**UPDATE #3:**
```
[ o.k. ] Building package [ linux-focal-root-edge-nanopi-r4s ]
[ o.k. ] Starting rootfs and image building process for [ edge nanopi-r4s focal   no ]
[ o.k. ] Checking for local cache [ focal-cli-arm64.e1e...b15.tar.lz4 ]
[ o.k. ] Extracting focal-cli-arm64.e1e...b15.tar.lz4 [ 0 days old ]
[ .... ] focal-cli-arm64.e1e...b15.tar.lz4:  350MiB [ 185MiB/s] [=========================================================================>] 100%
[ o.k. ] Adding Armbian repository and authentication key [ /etc/apt/sources.list.d/armbian.list ]
mount: /root/r4s/build/.tmp/rootfs-6945ec50-43aa-4ce1-9d3c-53eabc9c2ea9/dev: permission denied.
[ o.k. ] Applying distribution specific tweaks for [ focal ]
[ o.k. ] Applying common tweaks
[ .... ] Updating [ package lists ]

--------------> At this moment process "qemu-aarch64-static /usr/lib/apt/methods/http" hangs <--------------
--------------> At this moment sending SIGHUP to "qemu-aarch64-static /usr/lib/apt/methods/http" process <--------------

[ .... ] Temporarily disabling [ initramfs-tools hook for kernel ]
[ .... ] Installing [ linux-u-boot-edge-nanopi-r4s_21.05.0-trunk_arm64.deb ]
[ .... ] Installing [ linux-image-edge-rockchip64_21.05.0-trunk_arm64.deb ]
[ .... ] Installing [ linux-dtb-edge-rockchip64_21.05.0-trunk_arm64.deb ]
[ .... ] Installing [ armbian-firmware_21.05.0-trunk_all.deb ]
[ .... ] Installing [ armbian-config_21.05.0-trunk_all.deb ]
[ .... ] Installing [ armbian-zsh_21.05.0-trunk_all.deb ]
[ .... ] Installing from repository [ wireguard-tools --no-install-recommends ]

--------------> At this moment process "qemu-aarch64-static /usr/lib/apt/methods/http" hangs again <--------------
--------------> At this moment sending SIGHUP to "qemu-aarch64-static /usr/lib/apt/methods/http" process <--------------

[ error ] ERROR in function install_deb_chroot [ image-helpers.sh:148 ]
[ error ] Installation of wireguard-tools --no-install-recommends failed [ nanopi-r4s focal no rockchip64 ]
[ o.k. ] Process terminated
[ o.k. ] Unmounting [ /root/r4s/build/.tmp/rootfs-6945ec50-43aa-4ce1-9d3c-53eabc9c2ea9/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:65 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
root@armbian-builder:~/r4s/build#
```

starkovv, 2021-04-08:
> I use KVM where things works - compiling Focal image works on Focal host, while Qemu crashes if you want to build Hirsute or Bullseye target. For that, you need to use Hirsute host. That is also one option to follow up.

Does it mean that test builds named Hirsute published at https://www.armbian.com/nanopi-r4s build on Hirsute host? 🤔

igorpecovnik, 2021-04-08:
> build on Hirsute host?

Yes.

starkovv, 2021-04-08:
Yep. Hirsute made it.

```
[...]
[ o.k. ] Done building [ /home/vagrant/armbian/build/output/images/Armbian_21.05.0-trunk_Nanopi-r4s_focal_edge_5.11.12.img ]
[ o.k. ] Runtime [ 187 min ]
```

**Issue Solved.**

heisath, 2021-04-09:
Closing this as it has been solved. Hopefully qemu will fix these edge cases which keep popping up from time to time.

closed 2021-04-09 by heisath