armbian/build #2769: unable to build odroidxu4 u-boot

belegdol, 2021-04-13:
I am unable to build kernel for odroidxu4-current due to issues with u-boot:
```
	== u-boot make  ==

.config:927:warning: override: reassigning to symbol BOOTDELAY
.config:927:warning: symbol value '' invalid for BOOTDELAY
In file included from tools/../include/libfdt.h:54,
                 from tools/fdt_host.h:11,
                 from tools/imagetool.h:24,
                 from tools/atmelimage.c:8:
/usr/include/libfdt_env.h:27:30: error: conflicting types for ‘fdt64_t’
   27 | typedef uint64_t FDT_BITWISE fdt64_t;
      |                              ^~~~~~~
In file included from <command-line>:
././include/libfdt_env.h:19:16: note: previous declaration of ‘fdt64_t’ was here
   19 | typedef __be64 fdt64_t;
      |                ^~~~~~~
```
I tried adapting the patch from https://github.com/armbian/build/commit/757d9bc8d24eca1dea84a126289166c23e65a68e (the headers are in include subdir) but it did not help. Uninstalling libfdt-dev does not help because compile.sh installs it back.

EvilOlaf, 2021-04-13:
Do a `git pull` to make sure you are up to date with master.

Try removing here then uninstall: https://github.com/armbian/build/blob/5a91d9c4294bee4c39528ddf01922cfc163a2a2e/lib/general.sh#L1003

@igorpecovnik

belegdol, 2021-04-13:
@EvilOlaf, i did and I managed to get the build to complete. The issue report is to get it fixed for others.

EvilOlaf, 2021-04-13:
Ok then this libfdt needs further adjustments.

igorpecovnik, 2021-04-13:
This https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-imx6/libfdt_fix.patch needs to be added. Probably.

belegdol, 2021-04-13:
I tried, adapting the path to include as this is where the headers are. It did not help.

igorpecovnik, 2021-04-13:
Workaround should still work - uninstall libfdt-dev ... but i guess we need to solve this better. I already lost afternoon on this.

belegdol, 2021-04-13:
FYI I tried the following patch:
```patch
From e0d20dc1521e74b82dbd69be53a048847798a90a Mon Sep 17 00:00:00 2001
From: Paul Kocialkowski <contact@paulk.fr>
Date: Fri, 2 Mar 2018 23:13:42 +0100
Subject: [PATCH] tools: Include U-Boot libfdt headers from their actual path

There are no headers for libfdt in lib/libfdt, as they are instead
located in scripts/dtc/libfdt. Specifying lib/libfdt for headers
inclusion in host tools results in using the system libfdt headers,
which is not what we want. Change this to the proper path.

---
 tools/Makefile | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/tools/Makefile b/tools/Makefile
index d3387fad69..f38f68ee47 100644
--- a/tools/Makefile
+++ b/tools/Makefile
@@ -244,7 +244,7 @@ endif # !LOGO_BMP
 #
 HOST_EXTRACFLAGS += -include $(srctree)/include/compiler.h \
 		$(patsubst -I%,-idirafter%, $(filter -I%, $(UBOOTINCLUDE))) \
-		-I$(srctree)/lib/libfdt \
+		-I$(srctree)/include \
 		-I$(srctree)/tools \
 		-DUSE_HOSTCC \
 		-D__KERNEL_STRICT_NAMES \
-- 
GitLab
```
It does not seem to change anything:
```
	== u-boot make  ==

.config:927:warning: override: reassigning to symbol BOOTDELAY
.config:927:warning: symbol value '' invalid for BOOTDELAY
In file included from tools/../include/libfdt.h:54,
                 from tools/fdt_host.h:11,
                 from tools/imagetool.h:24,
                 from tools/aisimage.c:8:
/usr/include/libfdt_env.h:27:30: error: conflicting types for ‘fdt64_t’
   27 | typedef uint64_t FDT_BITWISE fdt64_t;
      |                              ^~~~~~~
In file included from <command-line>:
././include/libfdt_env.h:19:16: note: previous declaration of ‘fdt64_t’ was here
   19 | typedef __be64 fdt64_t;
      |                ^~~~~~~
```

closed 2021-04-13 by igorpecovnik