armbian/build #2777: [Security] Problem found in htop (your patches)

BenBE, 2021-04-16:
**Describe the bug**
Several security issues in your [downstream patches](https://github.com/armbian/build/tree/master/packages/extras-buildpkgs/htop/debian/patches) to `htop`.

**To reproduce**
Have a look at those patches and see how many instances of unchecked buffer writes they contain, potentially allowing for DoS or code execution. One such major offender is the [`avafinger-cpu-monitor.patch`](https://github.com/armbian/build/blob/master/packages/extras-buildpkgs/htop/debian/patches/avafinger-cpu-monitor.patch). Look for `strcpy`.

**Provide logs:**
Please contact upstream for full details.

lanefu, 2021-04-16:
Do you have any suggested remediations?

BenBE, 2021-04-16:
In the easiest case you should replace those `strcpy` calls by checked calls like `strncpy` to ensure no write beyond the end of the allocated buffers is possible. Recent versions of htop also include a function [`String_safeStrncpy` in `XUtils.c`](https://github.com/htop-dev/htop/blob/master/XUtils.c#L196), that further avoids one pitfal regarding plain use of `strncpy` alone.

Overall all externally provided strings should either be copied into dynamically managed buffers (`xStrdup`) or length-checked before further processing, when statically allocated buffers are used.

piter75, 2021-04-16:
@BenBE thanks for the heads-up!
The patches were copied over without proper attention :(
We will most probably borrow `String_safeStrncpy` from the current htop version to fix it.

PR #2778 (2021-05-02): Fix unchecked buffer writes in custom htop code
# Description

Fix unchecked buffer writes in copied custom htop code
Closes: #2777 

# How Has This Been Tested?

Please describe the tests that you ran to verify your changes. Please also note any relevant details for your test configuration.

- [x] built htop locally and run on one of the boards

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [ ] I have made corresponding changes to the docume [... cut]
closed 2021-05-02 by lanefu