armbian/build #2787: [SID/Bullseye] Debootstrap fails at second stage

baalajimaestro, 2021-04-21:
**Describe the bug**
Debootstrap fails when building debian>10.

**To Reproduce**
./compile.sh EXPERT="yes"
Choose Debian bullseye/sid and try compiling

**Provide logs:**
I didnt debug it more, but I know a workaround.

I tried googling about this issue, landed on this (https://forum.armbian.com/topic/16740-debootstrap-base-system-second-stage-failed/), I am sure its some dependency of library/something being broken on focal/buster.

**Workaround:**
Use hirutse host aka Ubuntu 21.04, the build seems to run fine, with no issues.

I am posting it here, to both notify the devs about the issue, and Ill be happy to help you with logs/anything if guided down on how to pull them up. I hope someone with similar issues lands here while googling about it

EvilOlaf, 2021-04-21:
Known limitation. As already figured out the workaround is use hirsute.

igorpecovnik, 2021-04-21:
Issue is related to Qemu where we can't do much about. In the past we also had issues with deboostrap, while the rest is less critical or fully under our control.

Best way would be to declare Ubuntu 21.04 as the only supported build host, but we had / have to wait for its official release. We already use it for producing daily builds for several months now.

psztoch, 2021-04-21:
You can use bullseye for your builder machine, too.
If you are Debian man, then you should use Debian Bullseye, as I. ;-)

lanefu, 2021-04-21:
FYI `20.10 (Groovy Gorilla)` is enough to get bullseye to build

PR #2790 (2021-04-22): Unlock Hirsute
# Description

Hirsute is due for release tomorrow. Also for building it or Sid / Bullseye we need it.

Resolves #2787 

Jira reference number [AR-732]

# How Has This Been Tested?

- [x] Build native hirsute
- [ ] Build under Docker (broken [due bug](https://stackoverflow.com/questions/66319610/gpg-error-in-ubuntu-21-04-after-second-apt-get-update-during-docker-build))

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in hard-to-understand areas
- [x] I have  [... cut]
closed 2021-04-22 by lanefu