armbian/build #2795: orangepi3 armbian-bsp-cli-orangepi3_21.08.0-trunk_arm64.deb issue

kasinjsh, 2021-04-23:
I'm trying to build from source orangepi3 images, but I cant build any image. After inspecting logs builds fails on multiple issues:

> Displaying message: Installation of /root/armbian-bsp-cli-orangepi3_21.08.0-trunk_arm64.deb failed orangepi3 focal yes sunxi64 err

> Displaying message: Installation of armbian-bsp-cli-orangepi3 failed orangepi3 hirsute yes sunxi64 err

> Displaying message: Extracting buster-xfce-arm64.afe...a22.tar.lz4 0 days old info
Displaying message: ERROR in function create_rootfs_cache debootstrap.sh:150 err
Displaying message: Cache /home/test/build/cache/rootfs/buster-xfce-arm64.afe5c1ecea6f661fe314db47ff9aaa22.tar.lz4 is corrupted and was deleted. Restart.  err

igorpecovnik, 2021-04-24:
How exactly do you build? We need to see complete builds steps. 

As you can see, our CI builds build this image https://imola.armbian.com/dl/orangepi3/nightly/

Shaggy013, 2021-04-24:
clean /home/test/build/cache/rootfs/ and let the build download a fresh one when running build script.

armbian script got a little confused sometimes with it if there are multiple rootfs in that folder.

igorpecovnik, 2021-04-24:
> and let the build download a fresh one when running build script

That is not "build from sources" :) More like an assembly from pre-build binaries.

Currently - for a week or so - this is broken since we changed package name and repository needs to be updated.

> armbian script got a little confused sometimes with it if there are multiple rootfs in that folder.

That is machine generated cache which don't need to be human readable .... with additional desktops, lots of new cache files are getting generated.

Shaggy013, 2021-04-24:
a week ? i dont wanna offend you but last 2 weeks i didnt build a armbian image.

 and this happens even 2 months ago.

kasinjsh, 2021-04-25:
I tested a bit more. Now it builds, I had issues if I use "Use precompiled packages from Armbian...".

igorpecovnik, 2021-04-25:
Yes, this is known to be broken and will automatically be fixed once we go out [with release update](https://forum.armbian.com/topic/17216-armbian-v2105-jerboa-release-thread/) ... closing.

closed 2021-04-25 by igorpecovnik