armbian/build #2798: armbianmonitor -u issue (maybe)

EvilOlaf, 2021-04-24:
https://forum.armbian.com/topic/17822-invalid-no-wpa2-in-access-point-mode-wifi-on-opi3-only-wpa/?do=findComment&comment=123399

`System diagnosis information will now be uploaded to gzip: /var/log/armbian-hardware-monitor.log.1.gz: No such file or directory`
I've seen this kind a error message now a couple of times in forums. What I found so far:

```
root@nas:~# bash -x armbianmonitor -u
+ Main -u
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ test -t 1
++ tput colors
+ ncolors=8
+ test -n 8
+ test 8 -ge 8
++ tput bold
+ BOLD=''
+ NC='\033[0m'
+ LGREEN='\033[1;32m'
+ LRED='\e[0;91m'
+ '[' 1 -eq 0 ']'
+ ParseOptions -u
+ getopts hHbBuUrRmMsnNd:Dc:C:pPvz c
+ case ${c} in
+ fping ix.io
+ grep -q alive
+ '[' 0 '!=' 0 ']'
+ which curl
+ echo -e 'System diagnosis information will now be uploaded to \c'
System diagnosis information will now be uploaded to + CollectSupportInfo
+ [[ -s /var/log/armbian-hardware-monitor.log ]]
+ zcat /var/log/armbian-hardware-monitor.log.1.gz
+ sed -E 's/([0-9]{1,3}\.)([0-9]{1,3}\.)([0-9]{1,3}\.)([0-9]{1,3})/XXX.XXX.\3\4/g'
+ curl -F 'f:1=<-' ix.io
gzip: /var/log/armbian-hardware-monitor.log.1.gz: No such file or directory
+ [[ -f /boot/armbianEnv.txt ]]
++ awk -F= '/^verbosity/ {print $2}' /boot/armbianEnv.txt
+ LOGLEVEL=1
+ '[' 1 -gt 4 ']'
+ lsusb
+ echo -e '\n### lsusb:\n'
+ lsusb
+ echo ''
+ lsusb -t
+ lspci
+ echo -e '\n### lspci:\n'
+ lspci
+ nvme
+ '[' -z ']'
[...]
```

igorpecovnik, 2021-04-24:
App should check if file exits and just abort if file not found - with "no data or similar"

AristoChen, 2021-06-16:
Not sure if this meets your expectation

```diff
diff --git a/packages/bsp/common/usr/bin/armbianmonitor b/packages/bsp/common/usr/bin/armbianmonitor
index 95fba8ecb..c42c8b45a 100755
--- a/packages/bsp/common/usr/bin/armbianmonitor
+++ b/packages/bsp/common/usr/bin/armbianmonitor
@@ -857,7 +857,8 @@ gwYNGjRo0KBBw1eEfwH4UoBHAKAAAA==" | base64 --decode | tar xzf -
 } # PatchRPiMonitor_for_sun8i

 CollectSupportInfo() {
-       [[ -s /var/log/armbian-hardware-monitor.log ]] && cat /var/log/armbian-hardware-monitor.log || zcat /var/log/armbian-hardware-monitor.log.1.gz
+       [[ -s /var/log/armbian-hardware-monitor.log ]] && cat /var/log/armbian-hardware-monitor.log || \
+       [[ -s /var/log/armbian-hardware-monitor.log.1.gz ]] && zcat /var/log/armbian-hardware-monitor.log.1.gz || echo "no data or similar"
        [[ -f /boot/armbianEnv.txt ]] && LOGLEVEL=$(awk -F'=' '/^verbosity/ {print $2}' /boot/armbianEnv.txt) || LOGLEVEL=1
        if [ ${LOGLEVEL} -gt 4 ]; then
                VERBOSE='-v'
```