armbian/build #2861: Marvell A3700-utils-marvell and mv-ddr-marvell.git now uses master branch

pali, 2021-05-29:
Latest version is now in `master` branch, see official TF-A build instructions:
https://trustedfirmware-a.readthedocs.io/en/latest/plat/marvell/armada/build.html#armada37x0-builds-require-installation-of-3-components

Please update build scripts to point to up-to-date software:
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/lib/main.sh#L427-L428

CC: @heaterC

lanefu, 2021-05-29:
thanks @pali

pali, 2021-05-30:
Also the latest version of binaries-marvell is in branch `binaries-marvell-armada-SDK10.0.1.0` (but binaries-marvell is not used on A37x0, only for A70x0/A80x0)
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/lib/main.sh#L429

___

And upstream TF-A is at https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L4-L6

___

And you should build **production** images (without DEBUG=1), not debug versions.
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L13
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L19
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L115

___

Then you also need to set correct paths, `/release/` instead of `/debug/`
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L94
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L99
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L114
https://github.com/armbian/build/blob/10eb5947ea6021e96b392e52eddc612316299822/config/sources/families/mvebu64.conf#L116

___

See also **example how to build production release of Marvell firmware image** at documentation https://trustedfirmware-a.readthedocs.io/en/latest/plat/marvell/armada/build.html which was tested for A3720 that is working fine.

Also for A3720 you could build and include CZ.NIC `wtmi_app.bin` firmware into `WTMI_IMG` (described in above _example how to build_) to enable Hardware Random Number Generator support.

lanefu, 2021-05-30:
[AR-775]

lanefu, 2021-07-04:
task noted in JIRA, but ebin is a CSC device and not a development priority.    Will accept PR's for this, but closing issue.

closed 2021-07-04 by lanefu

pali, 2021-10-03:
Well, you know what needs to be done, just replacing URLs and branch names to what is written in ticket.

lanefu, 2021-10-03:
Will accept PRs.