armbian/build #2873: After building the image, an infinite loop  Unmounting

150balbes, 2021-06-04:
After building the image, an infinite loop (last master)
Does anyone get the same problem ?

[ o.k. ] SHA256 calculating [ Armbian_21.08.0-trunk_Jetson-nano_focal_current_5.10.42_xfce_desktop.img ]
[ warn ] GPG signing skipped - no GPG_PASS [ Armbian_21.08.0-trunk_Jetson-nano_focal_current_5.10.42_xfce_desktop.img ]
[ .... ] Fingerprinting 
[ o.k. ] Done building [ /home/user/build/output/images/Armbian_21.08.0-trunk_Jetson-nano_focal_current_5.10.42_xfce_desktop.img ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236 ]
^C[ o.k. ] Unmounting [ /home/user/build/.tmp/rootfs-75959f5b-0c65-4f31-a872-8ebf62c92236/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:65 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated

150balbes, 2021-06-04:
Undoing this commit fixed the issue. Most likely the problem is in lines 82 86

https://github.com/armbian/build/commit/541dc716929435ec8d050ec8fd3c21bf865d2fb1

igorpecovnik, 2021-06-04:
> Most likely the problem is in lines 82 86

Can you double check and make a PR. Tnx.

BTW.
I am trying to put a proper CI pipeline https://github.com/igorpecovnik/build/actions/runs/906215431 where I had issues.

150balbes, 2021-06-04:
I will check and give an exact description of the place with the error (PR)

igorpecovnik, 2021-11-01:
Did you notice this again?

150balbes, 2021-11-02:
no, I didn't get this problem anymore.

closed 2021-11-02 by igorpecovnik