armbian/build #2881: error build u-boot to ARM

150balbes, 2021-06-07:
The last commits (I don't know exactly which ones yet) broke the native build on ARM  for u-boot.

igorpecovnik, 2021-06-07:
How can this be reproduced? I just tested building on aarcj64 host:

Building deb [ linux-u-boot-bananapi-current_21.08.0-trunk.44_armhf.deb ]

and it's fine.

150balbes, 2021-06-08:
You need to use RK. It uses rkbin via quemu.

igorpecovnik, 2021-06-08:
Aha @iav was changing something around here lately.

150balbes, 2021-06-08:
I found the error. These keys should not be used on ARM. The libraries that I specified earlier should be used.

not right
PKG_PREFIX="qemu-x86_64-static -L /usr/x86_64-linux-gnu "

right
PKG_PREFIX="qemu-x86_64-static  "

iav, 2021-06-08:
I add that libraries because got an errors without it. 
What target you choose to build to get that error, @150balbes? seems this switch should be set depend on binary intel tools, used for u-boot building.

150balbes, 2021-06-08:
On which system (host) do you collect ?

скорее всего вы не видели первичную инфу когда добавлялась поддержка нативной сборки. Для правильной работы нужно добавить две библиотеки в систему ARM. тогда ни какие дополнительные ключи сборки не нужны и все версии правильно собираеются на любом устройстве ARM.

https://github.com/150balbes/build/tree/armbian-tv/packages/bsp/common/lib64
https://github.com/150balbes/build/tree/armbian-tv/packages/bsp/common/lib/x86_64-linux-gnu

iav, 2021-06-08:
I add this option to use during building in docker builder. And, armbian build system oriented to "to do all that should be done". Get "default" clean (±, for sure) system, clone armbian build repo, run ./compile with root (sudo) privileges — and get things done.

Я гоняю тестовые сборки на ODroid-N2 with focal ubuntu, docker need to make ubuntu hirsute images.
Кроме того, я старался, чтобы можно было собрать имидж системы, начав с чистого образа ubuntu 20.04 на amazon aws or ocacle cloud, x64 or aarch64 architecture. То есть не делая никаких насроек, а именно 
1. создал образ
2. git clone armbian build repo
3. run build

150balbes, 2021-06-08:
запустите сборку station-p1 и получите ошибку

150balbes, 2021-06-08:
пакет libc6-amd64-cross ставится, но он не помогает

iav, 2021-06-08:
I complete run on my ODroid-N2, aarch64, ubuntu focal, armbian build from master `https://github.com/armbian/build`.
command line: `./compile.sh BOARD=station-p1`. no docker used.
image created.
I look into part of build log where uboot tools runs — see no errors:
```
  LD      tpl/u-boot-tpl
  LD      spl/u-boot-spl
  OBJCOPY tpl/u-boot-tpl-nodtb.bin
  CAT     tpl/u-boot-tpl-dtb.bin
  COPY    tpl/u-boot-tpl.bin
  OBJCOPY spl/u-boot-spl-nodtb.bin
  CAT     spl/u-boot-spl-dtb.bin
  COPY    spl/u-boot-spl.bin
Image Type:   Rockchip RK33 (SD/MMC) boot image
Init Data Size: 116736 bytes

 load addr is 0x200000!
pack input ./u-boot-dtb.bin 
pack file size: 849349 
crc = 0x5fd3a830
pack uboot.img success! 
out:trust.bin
merge success(trust.bin)
Image Type:   Rockchip RK33 (SPI) boot image
Init Data Size: 69632 bytes
Boot Data Size: 108544 bytes
[ o.k. ] Building deb [ linux-u-boot-station-p1-current_21.08.0-trunk_arm64.deb ]
[ o.k. ] Adjusting [ packaging ]
[ o.k. ] * applying general-packaging-5.10.y.patch 
[ o.k. ] Adding [ Kernel splash file ]
```
then — where need to look to see an error you reporting?

я попробовал собрать на своём одноплатнике, без докера, в образе ubuntu focal. На первый взгляд всё собралось, опробовать образ не могу, так как целевой железки у меня нет.
Скорее всего, я просто не туда смотрю.
Подскажите пожалуйста, куда и как смотреть, чтобы увидеть.

Кстати, одновременно появилась мысль: посмотрите пожалуйста, а не установлена ли у вас в системе перед сборкой переменная среды `QEMU_LD_PREFIX`? Она отвечает за то же самое.

150balbes, 2021-06-08:
нашёл причину, нужно убирать дополнительные библиотеки.

iav, 2021-06-08:
Убирать у вас или из армбиана?
Должен ли я ещё что-то сделать?

150balbes, 2021-06-09:
убирать в моём git
вам ни чего делать не нужно

iav, 2021-06-20:
Как там?

150balbes, 2021-06-20:
Сори, я не понял вопроса.

iav, 2021-06-20:
I used yesterday with the need to add to AARCH64 Builder packages GCC-ARM-NONE-EABI, GCC-Arm-Linux-Gnueabi - We needed to build Arm-Trusted-Firmware. Just in case I warn. 

Я тут вчера столкнулся с необходимостью добавить на aarch64 builder пакетов gcc-arm-none-eabi, gcc-arm-linux-gnueabi — понадобились для сборки arm-trusted-firmware. На всякий случай предупреждаю. #2909

closed 2021-10-17 by 150balbes