armbian/build #2882: IP address of customized network interface won't be displayed

AristoChen, 2021-06-08:
**Describe the bug**
When we log in every time, `/etc/update-motd.d/30-armbian-sysinfo` will show some info on the screen, including IP address.

If we rename the network interface to some other customized name, it is possible that it might not match the pattern written [here](https://github.com/armbian/build/blob/master/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo#L22), then we are not able to see the IP address when we log in or execute `/etc/update-motd.d/30-armbian-sysinfo` manually.

Not sure if it's a good idea to exclude `lo` and `dummy0`, and directly show IP address for all other nerwork interfaces?

lanefu, 2021-06-13:
> Not sure if it's a good idea to exclude lo and dummy0, and directly show IP address for all other nerwork interfaces?

I think that's a good idea. It's simpler to just exclude the normal unwanted things.   Feel free to make a PR and assign to me.

AristoChen, 2021-06-13:
Sure, working on it, will send a PR later

closed 2021-06-14 by lanefu