armbian/build #2904: rtl88x2bu/hal/hal_com.c:11513 rtw_lps_state_chk crash on odroidn2+ with patch

n2qcn, 2021-06-17:
**Describe the bug**
USB WiFi rtl88x2bu device generates stack trace in dmesg for odroidn2+

**To Reproduce**
every time

**Provide logs:**
 drivers/net/wireless/rtl88x2bu/hal/hal_com.c:11513 rtw_lps_state_chk+0x54/0x70 [88x2bu]
[]  rtw_lps_state_chk+0x54/0x70 [88x2bu]
[]  SetHwReg+0x648/0x670 [88x2bu]
[]  rtl8822b_sethwreg+0xc44/0xc90 [88x2bu]
[]  sethwreg+0x154/0x15c [88x2bu]
[]  rtw_hal_set_hwreg+0x14/0x20 [88x2bu]
[]  rtw_set_ps_mode+0xe4/0x24c [88x2bu]
[]  LPS_Leave+0xc4/0xcc [88x2bu]
[]  rtl8822b_phy_haldm_watchdog+0x188/0x1bc [88x2bu]
[]  rtw_hal_dm_watchdog+0x24/0x34 [88x2bu]
[]  rtw_dynamic_chk_wk_hdl+0x3c/0x64 [88x2bu]
[]  rtw_drvextra_cmd_hdl+0xa8/0x3d8 [88x2bu]
[]  rtw_cmd_thread+0x2b8/0x410 [88x2bu]
[]  kthread+0x154/0x158
[]  ret_from_fork+0x10/0x34

This looks like https://github.com/cilynx/rtl88x2bu/issues/133 which contains a patch to  initialize a declared struct https://github.com/cilynx/rtl88x2bu/commit/3a08c6f50107f75ec2b894e00db0c57c8fab6b5a

Is it better to change the 88x2bu branch or update patch/misc/wireless-realtek-88x2bu-5.12.patch ?

[ o.k. ] Adding [ Wireless drivers for Realtek 88x2bu chipsets branch:5.6.1_30362.20181109_COEX20180928-6a6a ]
[ o.k. ] Checking git sources [ rtl88x2bu 5.6.1_30362.20181109_COEX20180928-6a6a ]
[ o.k. ] * applying wireless-realtek-88x2bu-5.12.patch 

is what Armbian is building.

igorpecovnik, 2021-06-17:
If this is easily fixable, please provide a fix.

n2qcn, 2021-06-18:
That didn't fix it, and as pointed out, this isn't Armbian build code, so closing this issue and if I find a fix will commit upstream.

n2qcn, 2021-06-20:
stack traces stopped after upgrading from 5.6.1_30362.20181109_COEX20180928-6a6a to 5.8.7.1_35809.20191129_COEX20191120-7777 from https://github.com/cilynx/rtl88x2bu in build/lib/compilation-prepare.sh

n2qcn, 2021-08-05:
reopened with PR attached above.

closed 2021-08-06 by igorpecovnik