armbian/build #2905: Hirsute with kernel 5.12 - no start first wizard

150balbes, 2021-06-18:
I try to run the latest version of Hirsute with kernel 5.12 on P1, the system starts, but the primary configuration wizard does not turn on and the system is on authorization blocking all consoles. 
Is this a local problem for me or is someone observing the same behavior on other models ?

igorpecovnik, 2021-06-18:
You build one or you tried from the dl.armbian.com?

150balbes, 2021-06-18:
image dl.armbian.com

igorpecovnik, 2021-06-18:
I have tried Rockpi E, which was build [in the same batch](https://github.com/armbian/build/runs/2836837827?check_suite_focus=true), and its working normally. But that was via SSH.

150balbes, 2021-06-18:
I checked the latest version from 17 06 21 , on the monitor only the message "autologin root" and then does not work. The UART console doesn't work the same way. I checked through SSH, but there is no IP address (the system does not register on the DHCP server). I checked different dtbs, different media (SD USB). the result is the same, stopping at the root autologin.

150balbes, 2021-06-18:
I tried this SD card with DTB from Rockpi4 on NanoPC T4. The UART console is up and running, and the primary setup wizard is up and running. It looks like the reason is in the DTB for P1. I'll check his condition now.

150balbes, 2021-06-22:
Not clear behavior (same for P1 and M1). The image from the offsite for P1. I only replace the DTB setting. Everything starts (there is a normal primary autologin root and the launch of the primary configuration wizard). I use the same image on P1 (with DTB for P1), there is no autologin root, the agetty service does not work. If  remove this (agetty) service from the primary startup, the system starts, you can log in as ROOT from the console, and the primary setup (setup wizard) starts. I don't understand why the service doesn't work so selectively.

igorpecovnik, 2021-06-22:
> I don't understand why the service doesn't work so selectively.

No idea - I will try to catch it, but sadly don't have this particular hardware to check.

150balbes, 2021-06-22:
I have a few suggestions of the cause, I'm checking them now. 

p.s. but this is the first time I've encountered this behavior, the hardware either works or it doesn't. In this case, the hardware works, I tried to run agetty in already running systems, there are no problems, everything is worked out without errors.

150balbes, 2021-06-24:
After a number of checks, I found out that the LAN network does not work. I checked the official image for p1 Hirsute on Renegade rk3399, everything works correctly with dtb from P1. So the problem is exclusively in the equipment P1\M1. It seems that something has changed in the kernel network stack on 5.12, which leads to a breakdown of the LAN operation on P1\M1.

150balbes, 2021-06-24:
I built a new version with the kernel 5.12.13 without patches and EXTRAWIFI, everything starts and works without problems on P1. It remains to find the "culprits", patches or WIFI modules. :)

I will make PR corrections based on the results.

150balbes, 2021-06-24:
reason, external modules 8723. Are these modules used somewhere else besides P1\M1 ?

igorpecovnik, 2021-06-24:
> reason, external modules 8723. Are these modules used somewhere else besides P1\M1 ?

They are used in (some versions of) Rockpi E
https://github.com/armbian/build/pull/2060

150balbes, 2021-06-24:
I've been thinking about two possible fixes. But if there are more devices, then there is only one left, disable the built-in RTL driver that is not used in the kernel config, which creates a conflict.

igorpecovnik, 2021-06-24:
We can try that approach - I must have Rockpi E with this chip somewhere so I can try if that helps.

150balbes, 2021-06-24:
I will now check this option and send a PR for a test on Rockpi E.

150balbes, 2021-06-25:
The problem turned out to be in the 8723du module itself, it does not work correctly with the new 5.12 kernel. Need to make corrections. But I'm not sure I'll be able to fix it any time soon.

igorpecovnik, 2021-06-25:
OK. We can disable it for 5.12.y and record a bug to Jira - I'll check if i can do something about.

150balbes, 2021-06-25:
I checked the latest source branch v5. 13. 4. Everything works with it without errors, there is WiFi.

igorpecovnik, 2021-06-25:
Great! Then just changing this https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh#L589 ?

150balbes, 2021-06-25:
I have already done this in my GIT. The question is, does it make sense to do PR or can you immediately make changes to the MASTER ?

igorpecovnik, 2021-06-25:
Yes, make a PR. Just one board is affected and it will probably just work. To stay on a safe side, change branch for kernel 5.12 and up.

150balbes, 2021-06-25:
The rtl8723DS module has the same problem with the 5.12 kernel. But I didn't find any updated drivers.

igorpecovnik, 2021-06-28:
This can be closed then.

closed 2021-06-28 by igorpecovnik