armbian/build #2910: Building in Docker ends with losetup complaining about unavailable loop device

Mausy5043, 2021-06-20:
**Describe the bug**
Building using `docker` for a Nano Pi Neo stumbles just before the finish line.   
`losetup: cannot find an unused loop device: No such device`

**To Reproduce**
Just in case someone thinks this matters, I'm using `Arch Linux version 5.10.44-1-lts (linux-lts@archlinux) (gcc (GCC) 11.1.0, GNU ld (GNU Binutils) 2.36.1)` on top of Intel Core i3-8100T (quad core).   
The system is headless, so I login via SSH with a `sudo`-enabled account.   

```
$ docker version
Client:
 Version:           20.10.7
 API version:       1.41
 Go version:        go1.16.4
 Git commit:        f0df35096d
 Built:             Fri Jun  4 08:14:39 2021
 OS/Arch:           linux/amd64
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.7
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.16.4
  Git commit:       b0f5bc36fe
  Built:            Fri Jun  4 08:14:24 2021
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          v1.5.2
  GitCommit:        36cc874494a56a253cd181a1a685b44b58a2e34a.m
 runc:
  Version:          1.0.0-rc95
  GitCommit:        b9ee9c6314599f1b4a7f497e1f1f856fe433d3b7
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

I have downloaded the git repository and run `$ ./compile.sh docker`   
Options selected:
  1. Full OS image for flashing
  2. Do not change kernel configuration
  3. nanopineo
  4. (kernel) current
  5. (release) buster
  6. (type) standard image

The build then proceeds for several minutes and ends thusly (providing only the last few lines here; full logs are probably not relevant but can be found [in this gist](https://gist.github.com/Mausy5043/d7fb496914cafd766bd364f578b4d87b))

```
[ .... ] Installing [ linux-u-boot-nanopineo-current_21.08.0-trunk_armhf.deb ]
[ .... ] Installing [ linux-image-current-sunxi_21.08.0-trunk_armhf.deb ]
[ .... ] Installing [ linux-dtb-current-sunxi_21.08.0-trunk_armhf.deb ]
[ .... ] Installing [ armbian-bsp-cli-nanopineo_21.08.0-trunk_armhf.deb ]
[ .... ] Installing [ armbian-firmware_21.08.0-trunk_all.deb ]
[ .... ] Installing [ armbian-config_21.08.0-trunk_all.deb ]
[ .... ] Installing [ armbian-zsh_21.08.0-trunk_all.deb ]
[ .... ] Installing from repository [ wireguard-tools --no-install-recommends ]
[ o.k. ] Enabling serial console [ ttyS0 ]
[ o.k. ] Enabling serial console [ ttyGS0 ]
[ o.k. ] Building kernel splash logo [ buster ]
[ .... ] Installing extras-buildpkgs [  hostapd htop mmc-utils sunxi-tools ]
[ o.k. ] Calling image customization script [ customize-image.sh ]
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-e8b7d0a1-7e8b-4b4e-9db5-38a20ba65472 ]
[ o.k. ] Preparing image file for rootfs [ nanopineo buster ]
[ o.k. ] Current rootfs size [ 1077 MiB ]
[ o.k. ] Creating blank image for rootfs [ 1352 MiB ]
[ .... ] dd: 1.32GiB [ 153MiB/s] [=======================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
losetup: cannot find an unused loop device: No such device
[ error ] ERROR in function prepare_partitions [ debootstrap.sh:544 ]
[ error ] Unable to find free loop device
[ o.k. ] Process terminated
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-e8b7d0a1-7e8b-4b4e-9db5-38a20ba65472/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:65 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

igorpecovnik, 2021-06-20:
Check those lines
https://github.com/armbian/build/blob/master/config/templates/config-docker.conf#L32-L34
in your userpatches/config-docker folder.

Mausy5043, 2021-06-20:
@igorpecovnik Thanks I'll look into it next time this happens to me.

I noticed that there were no /dev/loop* devices whatsoever.
While I was puzzling over this I did:
```
$ losetup -f
/dev/loop0
```

And then there was a loop device 🥳 , so I tried to build again and that was succesfull.
I'm assuming this won't survive a reboot, so I'll enable that option next time I run into this to see it that could work.

closed 2021-06-20 by Mausy5043

igorpecovnik, 2021-06-20:
Still strange - loop devices should be created automatically by the build script. 

We'll monitor this.