armbian/build #2928: Cannot build native on aarch64 platform both aarch64 ubuntu and armbina

YLinXin, 2021-06-27:
**Describe the bug**
build failed on aarch64 ubuntu and armbian@S922X(X88 King)

**To Reproduce**
1.run aarch64 ubuntu in QEMU on M1 macOS(beautiful price and beat the most of family computer.)
2.follow the instruction from [Developer-Guide_Build-Preparation](https://docs.armbian.com/Developer-Guide_Build-Preparation/) and [Enable native arm/arm64 building](https://armbian.atlassian.net/browse/AR-457) ,it seems the native build on aarch64 ubuntu is finished.
3. do some local work:
1. download [gcc-arm-10.2-2020.11-aarch64-aarch64-none-elf.tar.xz](https://developer.arm.com/-/media/Files/downloads/gnu-a/10.2-2020.11/binrel/gcc-arm-10.2-2020.11-aarch64-aarch64-none-elf.tar.xz) from [developer.arm.com](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads) and unpack&put to /opt/
2. `export PATH="$PATH:/opt/gcc-arm-10.2-2020.11-aarch64-aarch64-none-elf/bin"`
3. change dir to `cache/sources/u-boot-odroid/odroidg12-v2015.01/include/linux` and `cp compiler-gcc4.h compiler-gcc10.h` then commit it to prevent init.
4. `sudo ./compile.sh  EXPERT=yes PROGRESS_DISPLAY=plain DOWNLOAD_MIRROR=china OFFLINE_WORK=yes KERNEL_ONLY=yes KERNEL_CONFIGURE=yes BOARD=odroidn2 BRANCH=current RELEASE=hirsute`

**Provide logs:**
stdout:
```➜  build git:(master) ✗ sudo ./compile.sh  EXPERT=yes PROGRESS_DISPLAY=plain DOWNLOAD_MIRROR=china OFFLINE_WORK=yes KERNEL_ONLY=yes KERNEL_CONFIGURE=yes BOARD=odroidn2 BRANCH=current RELEASE=hirsute
[ o.k. ] Using config file [ /home/ubuntu/Desktop/build/userpatches/config-example.conf ]
[ o.k. ] Command line: setting EXPERT to [ yes ]
[ o.k. ] Command line: setting PROGRESS_DISPLAY to [ plain ]
[ o.k. ] Command line: setting DOWNLOAD_MIRROR to [ china ]
[ o.k. ] Command line: setting OFFLINE_WORK to [ yes ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ yes ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ yes ]
[ o.k. ] Command line: setting BOARD to [ odroidn2 ]
[ o.k. ] Command line: setting BRANCH to [ current ]
[ o.k. ] Command line: setting RELEASE to [ hirsute ]

[ .... ] *  [ You are working offline. ]
[ .... ] *  [ Sources, time and host will not be checked ]

[ .... ] Deboostrap components main,universe 
[ .... ] Deboostrap packages locales gnupg ifupdown apt-utils ca-certificates bzip2 console-setup cpio cron dbus init initramfs-tools iputils-ping isc-dhcp-client kmod less libpam-systemd linux-base logrotate netbase netcat-openbsd rsyslog systemd sudo ucf udev whiptail wireless-regdb crda dmsetup rsync tzdata rng-tools fdisk 
[ .... ] Packages bc cpufrequtils device-tree-compiler fping fake-hwclock psmisc chrony parted dialog ncurses-term sysfsutils toilet figlet u-boot-tools usbutils openssh-server nocache debconf-utils python3-apt smartmontools bridge-utils build-essential fbset iw wpasupplicant sudo linux-base crda wireless-regdb unattended-upgrades console-setup unicode-data initramfs-tools ca-certificates expect iptables automake html2text bison flex libwrap0-dev libssl-dev libnl-3-dev libnl-genl-3-dev keyboard-configuration gnupg2 networkd-dispatcher man-db hping3 selinux-policy-default dkms python3-distutils python3-lib2to3 vnstat network-manager network-manager-openvpn wireless-tools lsof htop mmc-utils wget nano sysstat net-tools resolvconf iozone3 jq libcrack2 cracklib-runtime curl mc i2c-tools software-properties-common alsa-utils btrfs-progs dosfstools iotop stress screen ntfs-3g vim pciutils evtest pv libfuse2 libdigest-sha-perl libproc-processtable-perl aptitude dnsutils f3 haveged hdparm rfkill vlan bash-completion hostapd git ethtool unzip ifenslave libpam-systemd iperf3 nfs-common libnss-myhostname f2fs-tools avahi-autoipd iputils-arping qrencode sunxi-tools 
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ hirsute ]
[ o.k. ] Downloading sources 
[ o.k. ] Checking git sources [ u-boot-odroid odroidg12-v2015.01 ]
[ .... ]  Cleaning ....  [ 6 files ]
[ o.k. ] Checking git sources [ linux-mainline linux-5.10.y ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ sunxi-tools master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ rkbin-tools master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ marvell-tools A3700_utils-armada-18.12 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ marvell-ddr mv_ddr-armada-18.12 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ marvell-binaries binaries-marvell-armada-18.12 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ odroidc2-blobs master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ testing-reports master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ amlogic-boot-fip master ]
[ .... ] Up to date 
[ o.k. ] Cleaning /home/ubuntu/Desktop/build/output/debs for [ odroidn2 current ]
[ o.k. ] Cleaning [ u-boot-odroid/odroidg12-v2015.01 ]
[ o.k. ] Compiling u-boot [ 2015.01 ]
env: ‘aarch64-none-elf-gcc’: No such file or directory
[ o.k. ] Compiler version [ aarch64-none-elf-gcc  ]
[ .... ] Checking out to clean sources 
[ o.k. ] Cleaning [ u-boot-odroid/odroidg12-v2015.01 ]
[ o.k. ] Started patching process for [ u-boot meson64-odroidn2-current ]
[ o.k. ] Looking for user patches in [ userpatches/u-boot/u-boot-odroid ]
[ o.k. ] * [l][c] fix-boot-scripts-path.patch 
[ o.k. ] * [l][c] fix_build_for_recent_host-side_libfdt.patch 
[ o.k. ] * [l][c] load-from-ext4.patch 
  HOSTCC  scripts/basic/fixdep
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/conf.o
[... 546 lines cut ...]
../arch/arm/cpu/armv8/g12b/bl31_apis.c:71:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
   71 |   asm __volatile__("" : : : "memory");
      |   ^~~
cc1: all warnings being treated as errors
make[3]: *** [../scripts/Makefile.build:275: arch/arm/cpu/armv8/g12b/bl31_apis.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[1]: *** [/home/ubuntu/Desktop/build/cache/sources/u-boot-odroid/odroidg12-v2015.01/Makefile:1138: drivers/mmc] Error 2
../common/bootm.c: In function 'get_fdto_totalsize':
../common/bootm.c:333:37: error: initialization of 'long long unsigned int' from 'void *' makes integer from pointer without a cast [-Werror=int-conversion]
  333 |  unsigned long long dtbo_mem_addr = NULL;
      |                                     ^~~~
../common/bootm.c: In function 'do_fdt_overlay':
../common/bootm.c:350:37: error: initialization of 'long long unsigned int' from 'void *' makes integer from pointer without a cast [-Werror=int-conversion]
  350 |  unsigned long long dtbo_mem_addr = NULL;
      |                                     ^~~~
make[2]: *** [../scripts/Makefile.build:420: arch/arm/cpu/armv8/g12b] Error 2
make[1]: *** [/home/ubuntu/Desktop/build/cache/sources/u-boot-odroid/odroidg12-v2015.01/Makefile:1138: arch/arm/cpu/armv8] Error 2
cc1: all warnings being treated as errors
make[2]: *** [../scripts/Makefile.build:276: common/bootm.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/home/ubuntu/Desktop/build/cache/sources/u-boot-odroid/odroidg12-v2015.01/Makefile:1138: common] Error 2
make: *** [Makefile:147: sub-make] Error 2```

**Some Thought**
I cannot understand the whole process of this project now, but there is no 'arm64' and 'aarch64' if-else statement in code. I try it step by step to build but failed. It seems a lot of thing to do.

igorpecovnik, 2021-06-27:
> it seems the native build on aarch64 ubuntu is finished.

aarch64 support is experimental. Odroid legacy uboot will never be supported. Only after [this](https://github.com/armbian/build/pull/2264) gets resolved.

lanefu, 2021-06-27:
@YLinXin as Igor mentioned it does vary by device.  building an allwinner or rock chip device.

ex:

Here's me building armbain for pineh64-b from armbian vm on M1

[image]

YLinXin, 2021-06-28:
> @YLinXin as Igor mentioned it does vary by device. building an allwinner or rock chip device.
> 
> ex:
> 
> Here's me building armbain for pineh64-b from armbian vm on M1
> 
> [image]

closed 2021-06-28 by YLinXin

YLinXin, 2021-06-28:
> @YLinXin as Igor mentioned it does vary by device. building an allwinner or rock chip device.
> 
> ex:
> 
> Here's me building armbain for pineh64-b from armbian vm on M1
> 
> [image]

-_- |  maybe I just have the custom rom on my X88 King.

```
.config:1607:warning: override: reassigning to symbol BOOTDELAY
tools/imx8image.c: In function 'copy_file.constprop':
tools/imx8image.c:414:7: warning: 'write' reading 5 bytes from a region of size 4 [-Wstringop-overflow=]
  414 |   if (write(ifd, (char *)&zero, 4 - tail) != 4 - tail) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./tools/imagetool.h:21,
                 from include/imx8image.h:13,
                 from tools/imx8image.c:8:
/usr/include/unistd.h:367:16: note: in a call to function 'write' declared with attribute 'read_only (2, 3)'
  367 | extern ssize_t write (int __fd, const void *__buf, size_t __n) __wur
      |                ^~~~~
tools/imx8mimage.c: In function 'copy_file.constprop':
tools/imx8mimage.c:293:7: warning: 'write' reading 5 bytes from a region of size 4 [-Wstringop-overflow=]
  293 |   if (write(ifd, (char *)&zero, 4 - tail) != 4 - tail) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from tools/imagetool.h:21,
                 from tools/imx8mimage.c:9:
/usr/include/unistd.h:367:16: note: in a call to function 'write' declared with attribute 'read_only (2, 3)'
  367 | extern ssize_t write (int __fd, const void *__buf, size_t __n) __wur
      |                ^~~~~
Image 'main-section' is missing external blobs and is non-functional: scp

/binman/u-boot-sunxi-with-spl/fit/images/scp/scp:
   SCP firmware is required for system suspend, but is otherwise optional.
   Please read the section on SCP firmware in board/sunxi/README.sunxi64

Some images are invalid

        == kernel ==

drivers/net/wireless/Kconfig:57: can't open file "drivers/net/wireless/rtl8189fs/Kconfig"
make[2]: *** [scripts/kconfig/Makefile:71: syncconfig] Error 1
make[1]: *** [Makefile:605: syncconfig] Error 2
make: *** [Makefile:713: include/config/auto.conf.cmd] Error 2
ARM64# ./compile.sh BOARD=pineh64-b BRANCH=current RELEASE=focal BUILD_MINIMAL=no SKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTIMAGE=sha,gpg,img
```

igorpecovnik, 2021-06-28:
> maybe I just have the custom rom on my X88 King.

No, the problem is very old u-boot sources which needs right version of (cross) compiler. It is terrible enough that we need to have [many toolchains on x86 build host](https://github.com/armbian/build/blob/master/lib/general.sh#L1191-L1201). We have absolutely no plan to repeat this on aarch64. Not even you start to pay for maintaining you this tool.