armbian/build #2930: NFS root option builds without a kernel

Jamesits, 2021-06-28:
I'm not sure if NFS rootfs target is still in maintenance. If not, please tell me. 

What I've done:
```shell
./compile.sh  BOARD=nanopi-r2s BRANCH=current RELEASE=buster BUILD_MINIMAL=yes BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img ROOTFS_TYPE=nfs
```

Expected result:

There should be a kernel plus some dtbs in `output/images/Armbian_21.08.0-trunk_Nanopi-r2s_buster_current_5.10.46_minimal_nfsboot.img`

Actual result:
- `Image` symlink exists, but kernel itself is missing
- `config` and `System.map` exists
- `dtb-5.10.46-rockchip64` is empty (all directories intact, files missing)

igorpecovnik, 2021-06-28:
> I'm not sure if NFS rootfs target is still in maintenance. 

This build system is well maintained, but ofc there are issues here and there. We don't have unit testing since we can't afford to have them. So it has to be investigated, when time permits. I think you will find the problem much faster if you dig in.

Jamesits, 2021-06-28:
Sure, I'll try debugging a bit. I just need to make sure you guys still think NFS is a feature, since the doc says the FEL target is not maintained, and NFS is documented on the same page.

igorpecovnik, 2021-06-28:
> I just need to make sure you guys still think NFS is a feature

Personally I haven't used this in ages - can't say.

Jamesits, 2021-10-21:
build log relevant to this bug:

```
... after customize-image.sh execution ...

[ o.k. ] No longer needed packages [ purge ]
[ o.k. ] Unmounting [ /home/vsts/work/1/s/armbian-build/.tmp/rootfs-de8e559f-d585-4751-9719-acfd72dfb057 ]
[ o.k. ] Preparing image file for rootfs [ nanopi-r2s bullseye ]
[ o.k. ] Current rootfs size [ 700 MiB ]
[ o.k. ] Using user-defined image size [ 64 MiB ]
[ o.k. ] Creating blank image for rootfs [ 64 MiB ]
[ o.k. ] Creating partitions [ /boot: ext4 root: nfs ]
[ .... ] Creating /boot [ ext4 on /dev/loop3p1 ]
[ o.k. ] Creating rootfs archive [ rootfs.tgz ]
[ .... ] Copying files to [ /boot ]
find: ‘/home/vsts/work/1/s/armbian-build/.tmp/mount-de8e559f-d585-4751-9719-acfd72dfb057/lib/modules/’: No such file or directory
[ error ] ERROR in function update_initramfs [ debootstrap.sh:680 ]
[ error ] No kernel installed for the version [ 5.10.63 ]
[ o.k. ] Process terminated 
[ o.k. ] Unmounting [ /home/vsts/work/1/s/armbian-build/.tmp/rootfs-de8e559f-d585-4751-9719-acfd72dfb057/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:82 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
##[error]Bash exited with code '255'.
```

The other logs are identical comparing to an EXT4 build besides minor deviations (download speeds, etc.). Is there some option to enable more detailed logs? I've already set `PROGRESS_DISPLAY="plain"`.

iav, 2022-01-25:
I did some trace.
I see first we have complete system image in folder `$SDCARD`. 
Then we do a .tgz of it, excluding /boot.
then copy content of /boot into `$MOUNT`.  `$MOUNT` is a tiny image, 64 MB by default. That size not enough for modern kernel, I set in to 96 MB for a while, just to continue (`configuration.sh:68, FIXED_IMAGE_SIZE=`).
And then looking for folder with kernel version in name like *5.15.16* there, in tiny /boot image (`debootstrap.sh:792`).
Sure, it's not there. it stay in `$MOUNT` folder.
Then in `update_initramfs` runs some more actions, looking like it should be run on full image of system, not on tiny boot.

So, how should it work? 
Maybe, we need to run `update_initramfs` before step "Creating rootfs archive", and run it to full image, not to `/boot` tiiny FS?

e.g. move code 
```
	[[ -n $KERNELSOURCE ]] && {
		update_initramfs $MOUNT
	}
```
before line `if [[ $ROOTFS_TYPE != nfs ]]; then`, and run it as `update_initramfs $SDCARD'?

I newer use network boot before, then I can't "just do what I propose", and check will it work or not. Too many unknown parts.

iav, 2022-01-27:
@Jamesits can you test this?