armbian/build #2941: exclude sunxi-tools from Orange Pi 4 build

AndreiCherniaev, 2021-06-29:
I suggest exclude sunxi-tools from Orange Pi 4 build. Because it use RK3399.

I use
`./compile.sh  BOARD=orangepi4 BRANCH=current RELEASE=hirsute BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_ONLY=no KERNEL_CONFIGURE=yes DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base COMPRESS_OUTPUTIMAGE=sha,gpg,img `

My log is
...
[ .... ] Installing from repository [ wireguard-tools --no-install-recommends ]
[ o.k. ] Freezing Armbian packages [ orangepi4 ]
[ o.k. ] Enabling serial console [ ttyS2 ]
[ o.k. ] Building kernel splash logo [ hirsute ]
[ .... ] Installing extras-buildpkgs [  hostapd htop mmc-utils **sunxi-tools** ]
[ o.k. ] Calling image customization script [ customize-image.sh ]
[ o.k. ] Unmounting [ /home/a/Downloads/myArmbianGit/build/.tmp/rootfs-bed95457-9ec3-4f54-89f5-280e14b32b16 ]
[ o.k. ] Preparing image file for rootfs [ orangepi4 hirsute ]

AndreiCherniaev, 2021-07-02:
Trash packeges are include
build/config/cli/hirsute/main
where hirsute is your Ubuntu version name. Does armbian has tool to use package lists depending on the board. For my board Orange Pi 4B package **sunxi-tools** is trash.

@zhangn1985

igorpecovnik, 2021-07-02:
Agree it is a trash, but on the other hand it is non critical and not urgent matter, something that can easily wait. There are "1000 requests" already waiting for our sponsorship (here, [here](https://forum.armbian.com/forum/38-feature-requests/) and [here](https://armbian.atlassian.net/browse/AR)). Since we receive no support from you, since developing sw is our cost, don't expect any fast response. This is open source - if this micro problem bothers you badly, [just start](https://docs.armbian.com/Developer-Guide_Build-Preparation/) and send a merge request to the code. If you are busy like we are, try to hire someone. We would need to do the same to cover your wish.

igorpecovnik, 2021-07-02:
> Does armbian has tool to use package lists depending on the board.

Yes we do, but it was recently introduced functionality, which is badly documented and we haven't start to use properly yet.

AndreiCherniaev, 2021-07-02:
> > Does armbian has tool to use package lists depending on the board.
> 
> I think we do, but it was recently introduced functionality, which is badly documented and we haven't start to use properly yet.
Are you about https://github.com/armbian/build/issues/1837 ?

igorpecovnik, 2021-07-02:
> Are you about #1837 ?

That is discussion only. Nothing from that was implemented in a way described there. At least not intentionally.

Its a part of https://github.com/armbian/build/pull/2776

lanefu, 2021-07-05:
> [ .... ] Installing extras-buildpkgs [ hostapd htop mmc-utils **sunxi-tools** ]
> [ o.k. ] Calling image customization script [ customize-image.sh ]

As a workaround. you could uninstall sunni-tools and other packages you don't want via the [customize-image script](https://docs.armbian.com/Developer-Guide_User-Configurations/#user-provided-image-customization-script).

ThomasKaiser, 2021-09-03:
> I suggest exclude sunxi-tools from Orange Pi 4 build. Because it use RK3399

Never heard of `sunxi-fel` (part of `sunxi-tools`)?

`sunxi-tools` is part of every Armbian install to allow any SBC running Armbian (even those with an RK3399 at its heart) to flash the eMMC of any Allwinner SBC out there via USB. See https://github.com/ThomasKaiser/sunxi-armbian-flasher-osx for some ideas.

closed 2022-01-03 by igorpecovnik