armbian/build #2962: Why delete usb3 supporting in rk3328.dtsi

boyodyang, 2021-07-03:
5.10.47

linux-5.10.y/arch/arm64/boot/dts/rockchip/rk3328.dtsi

	usbdrd3: usb@ff600000 {
		compatible = "rockchip,rk3328-dwc3", "snps,dwc3";
		reg = <0x0 0xff600000 0x0 0x100000>;
		interrupts = <GIC_SPI 67 IRQ_TYPE_LEVEL_HIGH>;
		clocks = <&cru SCLK_USB3OTG_REF>, <&cru SCLK_USB3OTG_SUSPEND>,
			 <&cru ACLK_USB3OTG>;
		clock-names = "ref_clk", "suspend_clk",
			      "bus_clk";
		dr_mode = "otg";
		phy_type = "utmi_wide";
		snps,dis-del-phy-power-chg-quirk;
		snps,dis_enblslpm_quirk;
		snps,dis-tx-ipgap-linecheck-quirk;
		snps,dis-u2-freeclk-exists-quirk;
		snps,dis_u2_susphy_quirk;
		snps,dis_u3_susphy_quirk;
		status = "disabled";
	};

lanefu, 2021-07-03:
You would need to check the commit history of these patches to see if it was done by armbian or upstream

```
lane@billroyall:~/GIT/build/patch/kernel/rockchip64-current$ fgrep rk3328.dtsi *
add-board-nanopi-r2s.patch:+#include "rk3328.dtsi"
add-board-rk3328-roc-pc.patch:+#include "rk3328.dtsi"
add-board-rockpi-e.patch:+#include "rk3328.dtsi"
add-rk3328-usb3-phy-driver.patch:diff --git a/arch/arm64/boot/dts/rockchip/rk3328.dtsi b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
add-rk3328-usb3-phy-driver.patch:--- a/arch/arm64/boot/dts/rockchip/rk3328.dtsi
add-rk3328-usb3-phy-driver.patch:+++ b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
board-rk3328-roc-cc-dts-enable-dmc.patch: #include "rk3328.dtsi"
board-rk3328-roc-pc-dts-enable-dmc.patch: #include "rk3328.dtsi"
rk3328-audio-and-renegade-supplies.patch:diff --git a/arch/arm64/boot/dts/rockchip/rk3328.dtsi b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-audio-and-renegade-supplies.patch:--- a/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-audio-and-renegade-supplies.patch:+++ b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-dtsi-mmc-vdec-usb3-tweaks.patch:--- a/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-dtsi-mmc-vdec-usb3-tweaks.patch:+++ b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-dtsi-z-mali-opp-table.patch:diff --git a/arch/arm64/boot/dts/rockchip/rk3328.dtsi b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-dtsi-z-mali-opp-table.patch:--- a/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-dtsi-z-mali-opp-table.patch:+++ b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-fix-mali-node.patch:diff --git a/arch/arm64/boot/dts/rockchip/rk3328.dtsi b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-fix-mali-node.patch:--- a/arch/arm64/boot/dts/rockchip/rk3328.dtsi
rk3328-fix-mali-node.patch:+++ b/arch/arm64/boot/dts/rockchip/rk3328.dtsi
```

https://github.com/armbian/build/tree/master/patch/kernel/archive/rockchip64-5.10

boyodyang, 2021-07-03:
> You would need to check the commit history of these patches to see if it was done by armbian or upstream
>
> ```
> lane@billroyall:~/GIT/build/patch/kernel/rockchip64-current$ fgrep rk3328.dtsi * [...]

Thanks for ur answer!

closed 2021-07-03 by boyodyang