armbian/build #3035: Armbian-ramlog causes permission problem with apt-cacher-ng logrotate

menschel, 2021-07-19:
Hi,

I have an Odroid HC2 running 24/7 and I'm constantly getting emails from it about failed logrotate attempts.

```
error: failed to rename /var/log.hdd/apt-cacher-ng/apt-cacher.log to /var/log.hdd/apt-cacher-ng/apt-cacher.log.1: Permission denied
error: failed to rename /var/log.hdd/apt-cacher-ng/apt-cacher.err to /var/log.hdd/apt-cacher-ng/apt-cacher.err.1: Permission denied
truncate: missing file operand
Try 'truncate --help' for more information.
```

The corresponding logrotate file switches user to apt-cacher-ng before rotating which then hits the permission problem.

`/etc/logrotate.d/apt-cacher-ng`
```
/var/log.hdd/apt-cacher-ng/apt-cache*.log {
        su apt-cacher-ng apt-cacher-ng
```

To get rid of this, I first changed owner of `/var/log.hdd/apt-cacher-ng/` to apt-cacher-ng but it kept coming back again and again. I found out that some process is resetting the owner of that directory back to root together with all files in it.

As far as I could trace back the issue, armbian-ramlog seems to be the culprit.

It seems that whatever armbian-ramlog does with the `/var/log.hdd` directory, it breaks the permissions while doing so.

I'm curious if this issue is known and what I can do about it.

Thanks and Best Regards.

lanefu, 2021-07-19:
share an `armbianmontior -u`  link if possible

igorpecovnik, 2021-07-19:
I have a feeling that we saw this bug in the past and that was fixed. We had several fixes on this mechanism:

https://github.com/armbian/build/commits/master/packages/bsp/common/usr/lib/armbian/armbian-ramlog
https://github.com/armbian/build/commits/master/packages/bsp/common/usr/lib/armbian/armbian-truncate-logs

menschel, 2021-07-20:
> share an `armbianmontior -u` link if possible

`/var/log/armbian-hardware-monitor.log` is empty, so nope

igorpecovnik, 2021-07-20:
Then at least content of /etc/armbian* release files.

igorpecovnik, 2021-07-20:
Also linux-root* and armbian-bsp* package versions

menschel, 2021-07-20:
```
$ cat /etc/armbian-release 
# PLEASE DO NOT EDIT THIS FILE
BOARD=odroidxu4
BOARD_NAME="Odroid HC1"
BOARDFAMILY=odroidxu4
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=7223cca4
VERSION=5.90
LINUXFAMILY=odroidxu4
BRANCH=next
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage

$ cat /etc/armbian-image-release 
# PLEASE DO NOT EDIT THIS FILE
BOARD=odroidxu4
BOARD_NAME="Odroid XU4"
BOARDFAMILY=odroidxu4
VERSION=5.75
LINUXFAMILY=odroidxu4
BRANCH=next
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage
IMAGE_UUID=47c1b112-510b-48e3-8592-ec32f555189e
IMAGE_UUID=3e4aef01-5dcc-45d4-9edc-890d2bdf48c8
IMAGE_UUID=a348c638-b98d-44cc-803e-520861bc5f17

$ apt list --installed | grep root

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

linux-stretch-root-next-odroidxu4/now 5.90 armhf [installed,local]

$ apt list --installed | grep armbian

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

armbian-config/buster,now 21.05.6 all [installed]
armbian-firmware/buster,now 21.05.6 all [installed]
armbian-tools-stretch/now 5.75 armhf [installed,local]
hostapd/buster,now 3:2.9-102~armbian21.05.1+1 armhf [installed]
htop/buster,now 3.1.0-0~armbian20.08.2+1 armhf [installed]

```
which is interesting because `armbian-bsp-cli-odroidxu4` is not installed?!

igorpecovnik, 2021-07-20:
You are using some old board support packages for some reason - failed upgrade from previous packages versions. Dunno why exactly but not a build script issue.

We don't provide Stretch specific packages anymore so you will need to remove linux-root-* and manually install this package: https://imola.armbian.com/apt/pool/bullseye-utils/a/armbian-bsp-cli-odroidxu4/

menschel, 2021-07-20:
I should have seen it myself, that stretch package is an artifact from a previous release. That machine was upgraded to buster at some point by the usual `sed -e s/stretch/buster/g /etc/apt/sources.list ...` method

Actually I have 2 of those artifacts.

```
$ apt list --installed | grep stretch

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

armbian-tools-stretch/now 5.75 armhf [installed,local]
linux-stretch-root-next-odroidxu4/now 5.90 armhf [installed,local]

```
So I need to remove both stretch packages and then install
https://imola.armbian.com/apt/pool/buster-utils/a/armbian-bsp-cli-odroidxu4/

,right?

Can I break anything that way?

igorpecovnik, 2021-07-20:
- right.
- no.

menschel, 2021-07-21:
OK, I updated two odroid-hc2 that way and will observe if the issue comes up again. Thanks for the quick response.

```
$ cat /etc/armbian-release 
# PLEASE DO NOT EDIT THIS FILE
BOARD=odroidxu4
BOARD_NAME="Odroid XU4"
BOARDFAMILY=odroidxu4
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=428a20876-dirty
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=21.05.6
LINUXFAMILY=odroidxu4
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=Image
```

closed 2021-07-21 by igorpecovnik

menschel, 2021-07-26:
Just for the record. The permission errors have not popped up since a week. Thanks again.
It also seems to have solved an issue with avahi counting up the device name,
e.g. odroid-hc2-1 --> odroid-hc2-2