armbian/build #3066: orange pi3 uboot  build fail

vastchen, 2021-07-27:
when i want to build orangepi 3 

[ o.k. ] * [l][c] add-orangepi-zero-usb-boot-support.patch
[ o.k. ] * [l][c] add-orangepi-zeroplus2_h3.patch
[ warn ] * [l][c] add-pineh64-model-b.patch [ failed ]
[ o.k. ] * [l][c] add-sunvell-r69.patch
[ o.k. ] * [l][c] add-xx-boot-auto-dt-select-neo2.patch
[ o.k. ] * [l][c] add-xx-nanopi-k1-plus-emmc.patch
[ o.k. ] * [l][c] add-xx-nanopineocore2.patch
[ o.k. ] * [l][c] add-zeropi.patch
[ o.k. ] * [l][c] adjust-default-dram-clockspeeds.patch
[ o.k. ] * [l][c] adjust-small-boards-cpufreq.patch
[ warn ] * [l][c] enable-autoboot-keyed.patch [ failed ]
[ o.k. ] * [l][c] enable-ethernet-orangepiprime.patch
[ o.k. ] * [l][c] enable-nanopi-r1-uart1-console.patch
[ o.k. ] * [l][c] enable-r_pio-gpio-access-h3-h5.patch
[ o.k. ] * [l][c] fdt-setprop-fix-unaligned-access.patch

two patch fail 

'/tmp/tmp.fNvPDf0rDq/bl31.bin' -> './bl31.bin'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
sed: 无法读取 .config: 没有那个文件或目录      # just  Failed to read .config no the file 
scripts/kconfig/conf  --syncconfig Kconfig
[ error ] ERROR in function compile_uboot [ compilation.sh:241 ]
[ error ] U-boot compilation failed
[ o.k. ] Process terminated

vastchen, 2021-07-27:
thanks, got it .just wait fix? or git switch branch by hand?

igorpecovnik, 2021-07-27:
> just wait

Yes. Or DIY.

vastchen, 2021-07-27:
Thanks i got it.

igorpecovnik, 2021-07-27:
We will close this once its fixed. Now, this problem exists. Just we have no time to deal with.

igorpecovnik, 2021-08-16:
[Re-added patch for Orangepi 3.](https://github.com/armbian/build/commit/7f442070626ed8c53251cea6340f4aa65d645f2e#diff-68d5fbf7543aae8da81e70a1c9e6e64e1a61169f242b7397be9418fa82d593d1)

closed 2021-08-16 by igorpecovnik