armbian/build #3085: initramfs-cleanup requires update for initramfs-tools 0.138

MichaIng, 2021-08-07:
**Describe the bug**
The current [initramfs-cleanup](https://github.com/armbian/build/blob/master/packages/bsp/common/etc/kernel/preinst.d/initramfs-cleanup) hook script loops through installed initramfs versions found in the state directory `/var/lib/initramfs-tools`. Since v0.138 or v0.139, this directory is not used or updated anymore (the content is not used anymore for much longer), hence the cleanup is broken for Debian Bullseye, respectively Ubuntu Hirsute and above. Using this directory for Armbian's hook script was introduced with https://github.com/armbian/build/commit/a7e1fb1798c08c48bad73af5fb268170b1aef031, reasonably that time, and now this change can be moreless reverted. I'd personally suggest the following code line to be a bit more explicit about the file names, where the version string is expected to start after a dash:
```sh
find /boot -name 'initrd.img-*' -o -name 'uInitrd-*' ! -name "*-$version" -printf 'Removing obsolete file %f\n' -delete
```

For reference:
- https://lists.debian.org/debian-kernel/2020/09/msg00194.html
- https://metadata.ftp-master.debian.org/changelogs//main/i/initramfs-tools/initramfs-tools_0.140_changelog
- https://changelogs.ubuntu.com/changelogs/pool/main/i/initramfs-tools/initramfs-tools_0.139ubuntu3/changelog

**To Reproduce**
Steps to reproduce the behavior:
1. Flash a current Armbian Bullseye or Hirsute image
2. Upgrade the kernel, which triggers an initramfs update.
3. See error as of non-existent `/var/lib/initramfs-tools` and old `/boot/uIinitrd-<version>` remaining left afterwards.

igorpecovnik, 2021-08-08:
I have noticed some errors in this area but didn't get to digging-in part yet. Thanks. 

Also I think your propositions fits in.

The-going, 2022-02-24:
@igorpecovnik 
This is the same problem that is discussed [here](https://forum.armbian.com/topic/19707-hc1-kernel-update-gone-wrong/#comment-134829)
I have found a solution and am finishing testing.

MichaIng, 2022-02-24:
This while thing could be obsoleted by using `initrd.img-*` directly with modern U-Boot, which is cleaned by `initramfs-tools` itself, instead of `uInitrd-*`, obsoleting its build and cleanup scripts: https://armbian.atlassian.net/browse/AR-1080

The-going, 2022-02-24:
Misha, in fact, nothing needs to be added. Everything works using only the standard `initrd.img-$version`. But for the latest version of u-boot.
If you install the standard `u-boot-menu` package into the system and correctly configure the configuration file of this package in `/etc/default/u-boot`, you will be able to install and load various kernels of your choice.

But we need to keep using `uInitrd-$version` for backward compatibility, which means we need to fix the script to delete the generated files.

MichaIng, 2022-02-24:
Is `u-boot-menu` really required for this? I was thinking about the ability to simply replace `uIinitrd` with `initrd.img` in `/boot/boot.cmd`, creating a symlink to latest version (like done with uInitrd), or skipping it completely if U-Boot finds the right one automatically based on kernel version string (?). I tried that with latest "current" U-Boot on Odroid N2 without success.

In theory a U-Boot package upgrade could do the migration via maintainer script. However, until a decision about a migration has been done, for Bullseye a functional code line (supporting all distro versions) is given above 🙂.

The-going, 2022-02-24:
There is a difference in loading the kernel by u-boot itself and by another loader. If the kernel is loaded by u-boot itself, then it needs uInitrd. In the case of using the u-boot-menu package, u-boot does not load the kernel itself, but transfers control to the extlinux loader, and it already loads the kernel into the system. In this case, uInitrd is not needed. There are also some "rakes" related to the specifics of Armbian, which do not allow the standard initramfs-tools package to correctly delete unused files.

The-going, 2022-02-24:
> It removes `initrd.img-<version>` on individual kernel package removal

See `/etc/kernel/postrm.d/initramfs-tools`
```
# avoid running multiple times
if [ -n "$DEB_MAINT_PARAMS" ]; then
	eval set -- "$DEB_MAINT_PARAMS"
	if [ -z "$1" ] || [ "$1" != "remove" ]; then
		exit 0
	fi
fi
```
If you didn't explicitly specify 'remove' to remove the package, but only update it, nothing will be cleaned.
When are you updating the package `DEB_MAINT_PARAMS=upgrade`

MichaIng, 2022-02-24:
At that makes sense. Looks like I observed it wrong, likely when dealing with current, edge and legacy kernel packages where indeed one is removed instead of upgraded. The downside when not using a non-versioned meta package + versioned kernel package (like Debian) but a non-versioned kernel package. But it has upsides as well.

Back to the topic, instead of removing all non-matching initramfs versions before the kernel is installed/upgraded, another idea would be to remove those where no matching kernel image (or modules) is present after the kernel is installed, i.e. via postinst. That way it is generally possible to have multiple kernel versions installed, without having that preinst removing all other initramfs images. Also that way there is no time frame without any initramfs image anymore, which is a potential risk when the APT upgrade fails or the system crashes between preinst and postinst.

closed 2022-07-05 by MichaIng