armbian/build #3167: Any reason why jetson-nano-edge and jetson-nano-current kernel patch directories point to archive/imx6-5 ?

xlazom00, 2021-09-30:
**Describe the bug**
Any reason why jetson-nano-edge and jetson-nano-current kernel patch directories point to archive/imx6-5 ? It is pointing to totaly unrelated architecture directory

I think that armbian should add proper directory for jetson-nano architecture

igorpecovnik, 2021-09-30:
to include "general" patches, since there are none others.

closed 2021-09-30 by igorpecovnik

xlazom00, 2021-09-30:
@igorpecovnik I am not sure if any of this patches are general :)
https://github.com/armbian/build/tree/master/patch/kernel/archive/imx6-5.13

igorpecovnik, 2021-09-30:
I think making and empty folder or this solution takes identical time and both things are pointless since there is no value.

xlazom00, 2021-09-30:
@igorpecovnik empty foler make sense to me. As when anybody look they will se it is pure linux kernel

xlazom00, 2021-09-30:
@igorpecovnik I am making bugreport to this board to tegra kernel mailing list and it looks weird that armbian is applying unrelated patch set

igorpecovnik, 2021-09-30:
> weird that armbian is applying unrelated patch set

We never supported this hardware, we don't have anyone that would maintain this, so it is pretty irrelevant what is the state of this code. There were two people trying to do that ... but I can't possible do anything about :( 

You are welcome to take care of this board if you like.