armbian/build #3188: armbian-ramlog defaults break log rotation cron

lanefu, 2021-10-12:
```
root@odroidhc4:/home/lane# /usr/lib/armbian/armbian-truncate-logs
/usr/lib/armbian/armbian-truncate-logs: 18: /etc/default/armbian-ramlog: Syntax error: "(" unexpected
```

broken from #2894 

I thought we fixed this bug :( sorry everyone

lanefu, 2021-10-13:
ahh yeah this a Debian buster people

```
droidhc4:lane:# cat /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=odroidhc4
BOARD_NAME="Odroid HC4"
BOARDFAMILY=meson-sm1
BUILD_REPOSITORY_URL=git@github.com:lanefu/build.git
BUILD_REPOSITORY_COMMIT=b67ac420a
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=21.08.0-trunk
LINUXFAMILY=meson64
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
BRANCH=current
odroidhc4:lane:# head /usr/lib/armbian/armbian-truncate-logs
#!/bin/sh
#
# Copyright (c) Authors: https://www.armbian.com/authors
```

lanefu, 2021-10-13:
hmm also on bullseye 

```
ane@orangepiplus2e:~$ cat /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepiplus2e
BOARD_NAME="Orange Pi+ 2E"
BOARDFAMILY=sun8i
BUILD_REPOSITORY_URL=git@github.com:lanefu/build.git
BUILD_REPOSITORY_COMMIT=b67ac420a
DISTRIBUTION_CODENAME=bullseye
DISTRIBUTION_STATUS=supported
VERSION=21.08.0-trunk
LINUXFAMILY=sunxi
ARCH=arm
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=Image
BRANCH=current
lane@orangepiplus2e:~$ head /usr/lib/armbian/armbian-truncate-logs
#!/bin/sh
#
# Copyright (c) Authors: https://www.armbian.com/authors
#
```

useful64, 2022-01-13:
It is already fixed: https://github.com/armbian/build/pull/3115

closed 2022-01-13 by igorpecovnik