armbian/build #3298: AR-1009 or extensions framework + UEFI x86 + rpi4b + core changes/fixes ...

The-going, 2021-11-29:
### Description.
This discussion is intended so that we can formulate some points of the strategy for the development of the armbian build system.
In this case, we will be able to make the right choice, i.e. consistent, when we decide to change the code of the build system.

Let's continue this #3282

I want to express my gratitude to Ricardo Pardini @rpardini for the work done to expand the functionality of the build system.

Soon there will be a merge of this into the master branch and we need to provide (improve, change) the consistency of the change.

1) Development of the debian kernel package build script.
    Today we just copy two scripts to the source directory.
    This will be in the source package.
    
https://github.com/armbian/build/blob/1831d83acc9ef651f4f4d38eaccdbc8d2357c2b0/packages/armbian/builddeb#L291-L303

   How will this global variable be initialized when the kernel is built natively, without the armbian build system from the source package?

rpardini, 2021-11-29:
Hmm. I had _not_ considered this would be built outside the build system, but is indeed valid question.
Anyway, the same question could be asked of ALL the variables passed in https://github.com/armbian/build/blob/1831d83acc9ef651f4f4d38eaccdbc8d2357c2b0/lib/compilation.sh#L513-L530

I would much prefer if we used the standard builddeb from upstream kernel, since that does not do that symlink.

The-going, 2021-11-29:
Good health, Ricardo.
> Anyway, the same question could be asked of ALL the variables passed in

That's right, you're right. It is for this reason that the discussion has a general character.

> I would much prefer if we used the standard builddeb from upstream kernel, since that does not do that symlink.

[Standard scripts](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/scripts/package) from the kernel sources are very good and working examples\templates,
but this is not enough for industrial release.
Armbian has improved these scripts so that they do a little more.
My question is not how you handled the exception today and removed the creation of a symbolic link, but how we plan to do it in the future. I want to hear your opinion as an author.
I hope I wasn't too intrusive.

rpardini, 2021-11-29:
Thanks @The-going this is very valuable feedback. 
Unfortunately, doing reviews outside of the context of pull request is just too much work and I don't have the mental flexibility to maintain 3 different contexts.
Also the merge from my original PR into AR-1009 squashed my commits.
So I've taken the liberty to cherry pick, force push, and open #3300 draft from new AR-1009 branch into master.
I've moved your review there. I'd be very glad to continue the conversation there, please.

closed 2021-11-29 by rpardini