armbian/build #3339: e2fsprog 1.44.5-1+deb10u3 broken for >16tb partitions in armbian buster Odroid XU4 image

menschel, 2021-12-11:
**Describe the bug**

I'm using current Armbian Buster Odroid XU4 Image Kernel 5.4 on a Odroid HC-2 with Seagate EXOS X18 18TB HDD.
System has been updated to Bullseye.

```
$ apt list --installed | grep e2fsprogs lists version
e2fsprogs/oldstable,now 1.44.5-1+deb10u3 armhf [installed]
```

Apparently e2fsprog is build wrong on this image. It is not able to format an >16tb partition with ext4.
Instead it freezes after a few seconds at the point where the journal is being written. 
Therefore I run this via screen, then detach and ssh logout.

```
screen
sudo -i
mkfs.ext4 /dev/sda1 && shutdown now
CTRL+A
CTRL+D
CTRL+D
```

No progress seen even after a week. System runs forever on this task. Blue kernel activity led blinks frequently like on my other 3 machines.

**To Reproduce**
create >16TB partition, I use gdisk for that.

Try to format this partition with ext4
`mkfs.ext4 /dev/sda1`

**Provide logs:**
No logs available, systems does fill up RAM and stops responding on SSH and sometimes even on PING.
Power cycling the device is the only way to recover.

**FIX**
build e2fsprogs (e2fsprogs-1.46.4.tar.gz) yourself according to 
https://www.unix-ninja.com/p/Formatting_Ext4_volumes_beyond_the_16TB_limit 

after that it works, I have not tried without that auto_64bit flag.

igorpecovnik, 2021-12-11:
Thank you for your report but it doesn't look like as this would be a bug of the build framework (this repository) since correct version (Package: e2fsprogs (1.46.2-2)) is a part of the Bullseye user land: https://packages.debian.org/bullseye/e2fsprogs 

I would predict this is one of the bugs in the distribution upgrade which is prone to fail (upstream too), we don't recommend and is anyway not our work.

> I'm using 

Troubles using https://forum.armbian.com/

Closing.

Please reopen if my prediction doesn't stand (using freshly build image).

closed 2021-12-11 by igorpecovnik

menschel, 2021-12-11:
Ooops, I have to amend the info, that it was upgraded to bullseye. I updated that much Raspberrys to bullseye lately that I got confused the state of my machines about it.

That HC-2 is vanilla armbian buster and it was setup from scratch with the named image.
https://redirect.armbian.com/odroidxu4/Buster_current

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

but if this is actually a userland problem, who upstream do I contact about this issue?

Thanks for the quick reply.

menschel, 2021-12-11:
LOL, I should have bought the 16TB instead ;-P

I can't mount it, even found a similar problem described

https://serverfault.com/questions/462029/unable-to-mount-a-18tb-raid-6/536758#536758