armbian/build #3346: user can choise a image file for set in boot splash screen?

EmadDeve20, 2021-12-14:
# What is My Idea?
hello everybody so i working on a orangpi pluse2e and i need to create a boot splash screen for my customer!
my customer want use his company logo and i try to working about this (my attempt so far has not been successful :) )
so i think if we can create a option in ```compile.sh``` file for choise an image for setting boot splash screen can be *greate*!
what do you think? 

## How I can Do That?
I know about lpic 1 for Linux (more or less)
```i am stupid because i can`t create a boot splash screen right```? :D
I'm tried to search for this in the armbian forum but can not do it (create a custom boot splash screen)!

ok i know i need to learn more and so want to learn about this and i would like to contrib in this repository so my idea is good?

So what is my way and how can I fix it?(about this issue or my boot splash screen )

igorpecovnik, 2021-12-14:
> i know i need to learn more

To understand the background:
https://docs.armbian.com/User-Guide_FAQ/

> hello everybody so i working on a orangpi pluse2e and i need to create a boot splash screen for my customer!

This is not an issue of the build framework, but asking for support. Outside community support, those terms apply:
https://github.com/armbian/build#support -> https://forum.armbian.com/clients/donations/
Up to one week response time.

closed 2021-12-14 by igorpecovnik

EmadDeve20, 2021-12-14:
Thank You @igorpecovnik  about your help :D
 
have a good day

igorpecovnik, 2021-12-14:
> Thank You about your help

I wish I would like to say the same.

EmadDeve20, 2021-12-15:
hello @igorpecovnik  again i tried to change my boot splash screen you can help me?
[image]

but your link is not exist right now: https://github.com/armbian/build/blob/second/scripts/bootsplash

please i very need help! searched a some links but not working for me!
and this is another links:
https://forum.armbian.com/topic/10087-armbian-boot-splash-screen/
https://forum.armbian.com/topic/1416-how-to-change-the-boot-screen-in-the-assembly/
https://forum.armbian.com/topic/14197-like-initializing-boot-logo-image-aka-bootsplash/
https://forum.armbian.com/topic/15982-disable-boot-text-and-splash-screen/

i using a armbian without desktop and i install xfce desktop so i need create a boot splash screen but i know no how can i do this. this is plymouth is useful? i using server mode armbian without dekstop image and after install i installed desktop env becuase i think this can be low then image size and boot loader can work faster i right?

so let me see my project is d dashboard monitor so two topics is matter for me:
1 - fast boot 
2 - boot splash screen

i am so sorry for this question but i am so alone and i need help i can be so happy if you help me!
Thanks @igorpecovnik  and armenian community