armbian/build #3356: Kernel 5.10.87 update

gyohng, 2021-12-17:


igorpecovnik, 2021-12-17:
> would be to pin a specific kernel commit for armbian-build automatic checkout rather than checking out always the master branch tip of the mainline?

You mean temporally? Otherwise we would like to stick to the master.

If this change is a problem, perhaps we revert it with a patch. It would not be the first, nor the last time?

gyohng, 2021-12-17:
> You mean temporally? Otherwise we would like to stick to the master.
> If this change is a problem, perhaps we revert it with a patch. It would not be the first, nor the last time?

A patch would do perfectly well, I think. I assume it's not the goal to make a specific commit from the armbian-build repository result in a reproducible build for CI/CD?

Greg Kroah-Hartman's update in the mainline came an hour before I was doing the final image rebuild for the customer, and  I just accidentally spotted that the kernel version went up (even though I didn't pull anything from armbian), and it seemed to have this bug in it.

igorpecovnik, 2021-12-20:
Reverting that commit seems to be a solution. http://ix.io/3IVp

closed 2021-12-20 by igorpecovnik