armbian/build #3400: Allwinner platforms wrongly labeled as sun50iw1p1

ThomasKaiser, 2022-01-12:
Just FYI: https://github.com/armbian/build/blob/36c06fec2813ef127522a5e5c871a00cd2ad20b6/patch/kernel/archive/sunxi-5.15/general-arm64-add-proc-cpuinfo-entries.patch

This currently results in Allwinner platforms with `arm64` kernels such as `sun50iw2` (H5) and `sun50iw6` (H6) wrongly identifying itself as `sun50iw1p1`.

ThomasKaiser, 2022-01-12:
It all started to please userspace tools for the RPi: https://github.com/armbian/build/commit/5624db42123415d2620f893b2b4728a0e2a36a87

So RPi tools responsible for Armbian spreading misinformation and confusion with Allwinner SoCs. Guess what happens when running on a RPi with Ubuntu's kernel?

    root@rpi4b:/tmp# uname -a
    Linux rpi4b 5.15.7-bcm2711 #trunk.94 SMP PREEMPT Wed Dec 8 16:14:34 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
    root@rpi4b:/tmp# grep Hardware /proc/cpuinfo 
    root@rpi4b:/tmp# 

So 'Armbian on RPi' (LOL!) fails with some ancient GPIO libs while the original requirement of the very same GPIO lib that does some stupid parsing results in confusion with Allwinner SBC. 

I hope it's clear that the rogue patch can be deleted immediately to stop further confusion...

igorpecovnik, 2022-01-12:
> Guess what happens when running on a RPi with Ubuntu's kernel?

That's not Ubuntu kernel. That's https://github.com/raspberrypi/linux

> rogue patch can be deleted immediately 

That I do agree now.

igorpecovnik, 2022-01-12:
AFAIK Ubuntu kernel for Rpi is 5.4.y ..., recently moved to legacy branch. Further [changes are planned](https://github.com/armbian/build/pull/3391).

closed 2022-01-12 by igorpecovnik

igorpecovnik, 2022-01-12:
https://github.com/armbian/build/blob/master/config/sources/families/bcm2711.conf#L16-L34

> lsb_release -a

You are saying we should change how things are reported? 

> Of course also including the pathetic overvolting:

True, didn't notice. Not sure what we will do about that.

ThomasKaiser, 2022-01-12:
> You are saying we should change how things are reported?

I neither care about some reporting nor about the whole attempt to 'support' RPi. But maybe you should know that RPi kernels always contain the platform they're built for (`-v6`, `-v7+` or `-v8+`)

One of the better things of this platform is that it can run an ARMv7 userland since a 64-bit userland is just such an [insane waste of resources](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Review_of_RPi_Zero_2_W.md#armbian--ubuntu-jammy-on-zero-2).

> True, didn't notice. Not sure what we will do about that.

Removing of course. The DVFS on RPi is that pathetic that the CPU cores get fried all the time when `over_voltage` is set. 

Starting with silicon revision `C0` the BCM2711 SoCs default to 1.8 GHz when `arm_boost=1` is set in ThreadX (and with a more recent ThreadX version even with appropriate DVFS and lower voltage at lower clockspeeds). All this is gone when `over_voltage` is set. Especially if it's about a laughable increase of 200 MHz compared to the stock 1800 MHz (that work stably with `over_voltage=2` on `B0` [SoC revisions](https://github.com/ThomasKaiser/sbc-bench/blob/master/sbc-bench.sh#L1171-L1174)).

igorpecovnik, 2022-01-12:
> maybe you should know

Only 64b kernel / userland was planned. And no optimisations, improvements or competing with Raspbian or others unless drive will be coming in a Pull request form.

> is just such an insane waste of resources.

Well, software support is also huge waste of resources.

Overclocking will be disabled in one of the next fixes.

ThomasKaiser, 2022-01-12:
> Overclocking will be disabled in one of the next fixes.

The overvolting is the real problem. And I hope you and/or @rpardini remember the necessity to set `arm_boost=1`? This is required (combined with a current ThreadX on the FAT partition) to get the 1.8 GHz 'unlocking'. On `B0` SoCs (see above how to check) at first run `over_voltage=2` and `arm_freq=1800` could be set **inside the `[pi4]` section** to generate a somewhat consistent behaviour amongst different SoC/board revisions.

rpardini, 2022-01-13:
Hi @ThomasKaiser I've very innocently (I swear!) forgot that overvolt/overclock there and sincerely think it should be removed, especially before a formal release is done.