armbian/build #3405: packages are not created

ArgonautQ, 2022-01-16:
**Describe the bug**
while building a armbian system for a pine64 SBC (Sunxi Allwinner A64) the build process breaks after a while. I build on Ubuntu21.04 on a KVM/qemu virtual system.

**To Reproduce**
I checked out the repository changed to the build directory and invoked: ./compile.sh INSTALL_HEADERS=yes (to get kernel sources) 
Within kernel config I enabled kgdb, kernel with debugging information, and SCSI logging. Please look into config
https://nopaste.net/XirP7ZCLpR

**Provide logs:**
Drop logs to pastebin service of your choice 
[compilation.log](https://github.com/armbian/build/files/7877
[output.log](https://github.com/armbian/build/files/7877468/output.log)
465/compilation.log)
and supply logs from output/debug
[install.log](https://github.com/armbian/build/files/7877457/install.log)

The-going, 2022-01-16:
Please show the list of collected packages as:
```
 ls -al output/debs/
```
And the output of the diff command is between the default configuration file and your configuration file:
```
diff config/kernel/linux-sunxi64-current.config cache/sources/linux-mainline/LINUXDIR/.config >~/diff-configfile.patch
```

The-going, 2022-01-16:
@ArgonautQ I don't see your compilation log!

The-going, 2022-01-16:
Error: The patch did not apply:
board-h5-orangepi-zero-plus-fixes.patch

But this should not affect compilation. I started the check.

ArgonautQ, 2022-01-16:
[compilation.log](https://github.com/armbian/build/files/7877745/compilation.log)

The-going, 2022-01-16:
Ok!  i look

The-going, 2022-01-16:
Please archive the entire output/{debug,config} folder and the debian folder in the kernel sources.
The debian folder that is created during compilation.

 I'm going to have to sweat tomorrow. This is a very good reason to return to the kernel build script.

The-going, 2022-01-16:
I found a mistake. Thanks for the feedback.

The-going, 2022-01-17:
@ArgonautQ Packages are being collected. The image is being created. But you will have to install the kernel package for debugging yourself directly on the board. Please write here about the results of use.

closed 2022-01-22 by igorpecovnik

ArgonautQ, 2022-01-27:
tested with Hirsute Hippo and Bullseye: confirmed build with debugging kernel succeed.

The-going, 2022-01-27:
Thanks for the feedback!