armbian/build #3414: modules to include in Initrd

150balbes, 2022-01-20:
Is there an option (variable) in the build system that can set which modules to include in Initrd?

EvilOlaf, 2022-01-20:
None I am aware of...

150balbes, 2022-01-20:
I just found a problem in the current u-boot for Jetson Nano, which manifests itself in destroying the Initrd if it is larger than a certain size (14Mb), due to the too small range of addresses that are allocated for it. Loading the kernel falls into the Initrd range and spoils it. Therefore, as a temporary solution, I want to limit the size of the Initrd by excluding modules that are unnecessary to run from it. You can change the kernel config and remove some modules, but this solution worsens the capabilities of the universal kernel.

@rpardini in your advanced settings, is there such a possibility to define\describe the list of modules that should be used for Initrd ?

rpardini, 2022-01-22:
Hi Oleg, I've not yet touched much in that sense, `initramfs-tools` does it's own super-slow "heuristics" to determine the set of modules to include in the initrd, and I think Armbian just goes with that. It's mostly focused on platforms without initrd limits, and tries to include every storage, filesystem, network, etc driver/modules it finds. We can add a way to tweak those, I think, some initrd's are gigantic since our kernels have almost all modules always enabled, and building the initrd is slow and result is very big. Take a look at the Debian docs...

rpardini, 2022-01-22:
@150balbes  Take a look at `/etc/initramfs-tools/initramfs.conf`, things such as `MODULES=most` can be changed to `MODULES=list` and then set the ordered list of modules in `/etc/initramfs-tools/modules`; that will stop it from using the heuristics and force it to use the list you specify.

150balbes, 2022-01-22:
> I think, some initrd's are gigantic since our kernels have almost all modules always enabled, and building the initrd is slow and result is very big.

I found the cause of the problem on Jetson Nano and the solution for it. But I agree with you, it will be good to finalize the process of creating Initrd, now it includes everything as it is and often it does more harm than good. This will become especially relevant with the transition to universal cores, which will be used on several platforms at once. There will be a lot of modules in them, and if you include all the modules found, this may create the problem of too large Initrd.

> Take a look at `/etc/initramfs-tools/initramfs.conf`, things such as `MODULES=most` can be changed to `MODULES=list` and then set the ordered list of modules in `/etc/initramfs-tools/modules`; that will stop it from using the heuristics and force it to use the list you specify.

thanks, I'll take a look at this and try to check out different options for myself.

rpardini, 2022-02-05:
Multiple PR's merged regarding this, let's close this one.

closed 2022-02-05 by rpardini