armbian/build #3415: aptly.info expired

duke-, 2022-01-20:
Hello,

I'm unable to build a new image using docker since the aptly.info domain is expired (see: http//aptly.info & https://github.com/aptly-dev/aptly/issues/1009).

```
 ---> b1cea67b0ca2
Step 4/11 : RUN sh -c "     if [ $(dpkg --print-architecture) = amd64 ]; then         if [ x'' != x$http_proxy ]; then             curl -fsSL -x $http_proxy https://www.aptly.info/pubkey.txt | apt-key add - >/dev/null 2>&1;         else             curl -fsSL https://www.aptly.info/pubkey.txt | apt-key add - >/dev/null 2>&1;         fi;         echo \"deb http://repo.aptly.info/ nightly main\" > /etc/apt/sources.list.d/aptly.list;         dpkg --add-architecture i386         && apt-get update         && apt-get install -y --no-install-recommends         lib32ncurses6         lib32stdc++6         lib32tinfo6         libc6-i386         zlib1g:i386;     fi"
 ---> Running in cc214bc64994
curl: (28) Failed to connect to www.aptly.info port 443: Connection timed out
```

Is there any way to get around?

-- Micha

igorpecovnik, 2022-01-20:
I am adding package to our repositories. Will take a day to synch up.

closed 2022-01-20 by igorpecovnik

igorpecovnik, 2022-01-20:
I think it's safe to remove that package. Will be sure after whole CI run.